isaac-sim/IsaacLab #1298: [Bug Report] AttributeError: 'DeformableObject' object has no attribute '_data'. Did you mean: 'data'?

sabinakaminska95, 2024-10-24:
When I trying to add different deformable object by adding .stl model and after this trying to run ./isaaclab.sh -p source/standalone/environments/state_machine/lift_teddy_bear.py but with changing .usd with my own object the error is occur:

AttributeError: 'DeformableObject' object has no attribute '_data'. Did you mean: 'data'?

chenzyn, 2024-10-24:
I got the same error when I call the `write_root_pose_to_sim` method in 'RigidObject' Class.
```
AttributeError: 'RigidObject' object has no attribute '_data'. Did you mean: 'data'?
```

Mayankm96, 2024-10-24:
Usually the error happened before this popped up. Please check above in the terminal and see what other errors happen before this one.

RandomOakForest, 2024-11-22:
It seems you may have resolved this issue. Will close it for now. Please post a full error log may we have to investigate this further.

closed 2024-11-22 by RandomOakForest