isaac-sim/IsaacLab #1299: [Question] Pre-assembled Franka Panda + Robotiq gripper provided ? Issues with self assembled Franka panda + Robotiq gripper.

johnny-wang16, 2024-10-24:
### Question

I'm new to Isaaclab and was wondering if there are any pre-assembled files provided for Franka Panda + Robotiq 2f 85 gripper? I’ve seen the tutorials on rigging robots with robotiq gripper but it seemed out of date and unclear. I’ve also found broken robotiq 2f85 on omniverse server which is not that useful. I just need this robot configuration for reinforcement learning training. Please advise. Thanks!

To fix the gripper, I follow [this instruction](https://medium.com/@joolab/how-to-set-up-closed-loop-gripper-with-robotiq-2f-85-f7aac12936b2) and this [youtube video](https://www.youtube.com/watch?v=UpGgq-BR0Gg). The resulting gripper I got is [here ](https://drive.google.com/drive/folders/14_fAtWBHNLrhecq83eo7uYQ4jIL7SdMb?usp=sharing)and it seemed functional when applying force on to it. The franka file I used it the "franka_instanceable.usd" in [here](https://drive.google.com/drive/folders/11gtNh-YPGWQTIkGm4oFt3FxN_C3x-bes?usp=sharing).

Now that the gripper seemed fixed, I tried to connect them together with robot assembler tool provided in Isaac Utils but it does not work. The robot just flies everywhere for some reason.

Finally, I followed [this tutorial](https://medium.com/@joolab/how-to-assemble-a-gripper-on-a-robot-manipulator-in-isaac-sim-f7ba8e6f18fc) on assembling gripper on arm. Some details are not clear so I put my steps here:
1) "Open" the  `franka_instanceable.usd` so that I'll be able to delete the fingers and hand (instead of deactivate).

2) Drag and drop `2f85_instanceable.usd` [in this folder](https://drive.google.com/drive/folders/14_fAtWBHNLrhecq83eo7uYQ4jIL7SdMb?usp=drive_link) to "Layer". So layer structure looks like the image below. I think this allows me to delete the root_joint later on and change the transform of the gripper but im not sure if this is correct.

3) - 5) Follow the 3 steps in  [this tutorial](https://medium.com/@joolab/how-to-assemble-a-gripper-on-a-robot-manipulator-in-isaac-sim-f7ba8e6f18fc). 

6) I set damping and stiffness of the joints on robotiq gripper to 10 just for testing.

After all these steps, I tried running it but still faced several problems:
1) The grippers not aligned to the orientation I wanted. I changed the z to -45 and it seemed working when simulation is not running. But it immediately becomes misaligned when simulation starts. [Video here](https://drive.google.com/file/d/1tT30qESbWzxJKrI60Z3YSizm7LmBqhM3/view?usp=sharing).

2) The gripper acts weird when simulation starts. It first tries to close and eventually opens up. This is not desirable. This behavior can be seen from the video in 1). I suspect it might be due to the damping and stiffness parameter but not sure. Or maybe it's some pre-programmed trajectory that I'm aware of. 

3) Eventually in the reinforcement learning environment, I want to only use one value (0: close or 1: open) to indicate the action to the gripper. But currently, the gripper is controlled by drives in two joints (`finger_joint `and `right_outer_knuckle_joint`) and I'm not sure how to achieve what I want eventually in the reinforcement learning environment.

4) It's also unclear to me how I would import this in DirectRL environment because it is not a tree-like structure where the gripper nest under the final link of Franka. I tried to put it under link7 of franka but it causes more problems in simulation.

[image]
Here's my [assembled USD file](https://drive.google.com/drive/folders/1_gGSOMz2kgIiRuqCVTK_0FH6XhATL1P-?usp=sharing). Any advice would be greatly appreciated. Thank you very much for reading my problem.

jsmith-bdai, 2024-10-28:
First of all wanted to applaud your resourcefulness - glad you found a way to build up the Franka + 2F85 gripper!

Seems like you have resolved 1., so I can comment on 2. 

Your action space shouldn't be directly determined from the Articulation definition - this is specified in your `ActionCfg`. For the Cartpole example (https://isaac-sim.github.io/IsaacLab/main/source/tutorials/03_envs/create_manager_base_env.html#defining-actions) you'll see that while the cartpole has 2 joints, only the `slider_to_cart` joint is controlled. You can similarly configure your `ActionCfg` to only control those joints that you wish. 

Can you share your `EnvCfg` so I can take a look and make suggestions? I have a feeling you might be using `*` wildcard to select all joints for you action space.

johnny-wang16, 2024-10-28:
Thanks for your response!
This is my config:

@configclass
class FrankaCustomEnvCfg(DirectRLEnvCfg):
    # env
    episode_length_s = 8.3333  # 500 timesteps
    decimation = 2
    num_actions = 13
    num_observations = 23
    num_states = 0

    # simulation
    sim: SimulationCfg = SimulationCfg(
        dt=1 / 120,
        render_interval=decimation,
        disable_contact_processing=True,
        physics_material=sim_utils.RigidBodyMaterialCfg(
            friction_combine_mode="multiply",
            restitution_combine_mode="multiply",
            static_friction=1.0,
            dynamic_friction=1.0,
            restitution=0.0,
        ),
    )

    # scene
    scene: InteractiveSceneCfg = InteractiveSceneCfg(num_envs=4096, env_spacing=3.0, replicate_physics=True)

    # robot
    robot = ArticulationCfg(
        prim_path="/World/envs/env_.*/Robot",
        spawn=sim_utils.UsdFileCfg(
            usd_path=f"./Collected_franka_robotiq/franka_robotiq.usd",
            activate_contact_sensors=False,
            rigid_props=sim_utils.RigidBodyPropertiesCfg(
                disable_gravity=False,
                max_depenetration_velocity=5.0,
            ),
            articulation_props=sim_utils.ArticulationRootPropertiesCfg(
                enabled_self_collisions=False, solver_position_iteration_count=12, solver_velocity_iteration_count=1
            ),
        ),
        init_state=ArticulationCfg.InitialStateCfg(
            joint_pos={
                "panda_joint1": 1.157,
                "panda_joint2": -1.066,
                "panda_joint3": -0.155,
                "panda_joint4": -2.239,
                "panda_joint5": -1.841,
                "panda_joint6": 1.003,
                "panda_joint7": 0.469,
                # "panda_finger_joint.*": 0.035,
                "left_outer_knuckle_joint": 0.035,
                "right_outer_knuckle_joint": 0.035,
            },
            pos=(1.0, 0.0, 0.0),
            rot=(0.0, 0.0, 0.0, 1.0),
        ),
        actuators={
            "panda_shoulder": ImplicitActuatorCfg(
                joint_names_expr=["panda_joint[1-4]"],
                effort_limit=87.0,
                velocity_limit=2.175,
                stiffness=200.0,
                damping=4.0,
            ),
            "panda_forearm": ImplicitActuatorCfg(
                joint_names_expr=["panda_joint[5-7]"],
                effort_limit=12.0,
                velocity_limit=2.61,
                stiffness=200.0,
                damping=4.0,
            ),
            "robotiq_hand": ImplicitActuatorCfg(
                joint_names_expr=["left_outer_knuckle_joint", "right_outer_knuckle_joint"],
                effort_limit=200.0,
                velocity_limit=0.2,
                stiffness=2e3,
                damping=1e2,
            ),
        },
    )

jsmith-bdai, 2024-10-29:
Gotcha - didn't realize you were in Direct workflow, my bad. I re-read your question and realize you just want to be able to acquire non-articulated joints from `joint_names`?

The way we've defined joint names is to represent all of the joints (regardless of whether they are actuated). I think a workaround for you would be to hardcode the actuated joints e.g. `["left_outer_knuckle_joint", "right_outer_knuckle_joint"]` I can also check with the reset of the dev team to see if it makes sense to implement an `actuated_joint_names` property.

johnny-wang16, 2024-10-30:
Yes, i just want to be able to acquire non-articulated joints from joint_names. It's honestly not a big problem but i personally feel there should be a way to directly access the actuated ones. Otherwise, to set the joint positions using the [set_joint_position_target](https://isaac-sim.github.io/IsaacLab/main/source/api/lab/omni.isaac.lab.assets.html#articulation) method, I'll have to pass in some dummy values for the unactuated joints which is non ideal.

RandomOakForest, 2024-11-22:
Thanks for posting this. Great work! I would encourage you to try the latest release. I will tag this for review by the team.

HariKrishnan06082k, 2024-11-25:
Hi @RandomOakForest and @johnny-wang16 I'm having a similar problem in assembling the UR5e robot with Robotiq 2F85 gripper, I initially tried the robot assembler which broke the simulation when trying to enable the option of articulating under a single robot. I manually made the changes to gripper separately following (rigging complex structure tutorial) and tried creating a fixed joint from my robot tool 0 frame to robotiq gripper base. But I'm unable to articulate the entire robot properly as if i exert any force more than 10 N on gripper angular drives, my joints are impacted and they start rotating like crazy. I saw your franka usd file and it seems you are articulating it by 2 angular drives on inner knuckles, I'm trying to make it a interactive force driven grasp with mimic joints for the actuating drives (in my case finger_joint ) please suggest some fix to make the simulation more stable. Thanks. Attaching my USD for reference.

HariKrishnan06082k, 2024-11-25:
Please refer to following issues as well. 
[https://forums.developer.nvidia.com/t/robot-assembler-does-not-work-for-ur5e-and-robotiq-2f-85/301108](url)

StafaH, 2026-09-25:
Thanks @johnny-wang16. Isaac Sim now provides a preassembled Franka Panda with a Robotiq 2F-85 gripper as the `Robotiq_2F_85` variant of `FrankaRobotics/FrankaPanda/franka.usd`. Isaac Lab provides `FRANKA_ROBOTIQ_GRIPPER_CFG` for it in `source/isaaclab_assets/isaaclab_assets/robots/franka.py` (added in https://github.com/isaac-sim/IsaacLab/pull/3229). For DirectRL, select the driven joints with `find_joints(...)` and pass their IDs when setting position targets; `joint_names` includes passive joints.

closed 2026-09-25 by StafaH