isaac-sim/IsaacLab #1307: [Question] Where are the assets located?

meghbhalerao, 2024-10-25:
### Question

Hi, I am unable to find the xml or urdf files for the assets, for example cube, table etc. I see that the urdf files are located in a path which involves variables such as `NUCLEUS_ASSET_ROOT_DIR`  and `NVIDIA_NUCLEUS_DIR` but this seems to be on an NVIDIA server. 
Is there some place in the code which has a collection of all assets, i.e the `xml` files or the `urdf` or `mjcf` files? Thanks!

jsmith-bdai, 2024-10-28:
Most of the robots used within Isaac Lab rely on USD files that are all hosted on the Nvidia Nucleus server.

You can use `wget` to download them. As an example for Cartpole (on Isaac Lab 1.2):
```
wget http://omniverse-content-production.s3-us-west-2.amazonaws.com/Assets/Isaac/4.2/Isaac/IsaacLab/Robots/Classic/Cartpole/cartpole.usd
```

You can determine the `/Robots/Classic/Cartpole/cartpole.usd` bit from the robot's ArticulationCfg within `lab_assets`, e.g. [here](https://github.com/isaac-sim/IsaacLab/blob/9cc298e938c8903f4ba3d50a222c9c1f3a76c435/source/extensions/omni.isaac.lab_assets/omni/isaac/lab_assets/cartpole.py#L20)

jsmith-bdai, 2024-10-28:
I'll leave this open in case there's an easier way to do this

meghbhalerao, 2024-10-28:
Thanks!
Could this be added somewhere in the documentation, for example, potentially here https://isaac-sim.github.io/IsaacLab/main/source/how-to/import_new_asset.html, since I don't think I could find this info in the docs - 
[image]
Thanks again for your help!

RandomOakForest, 2024-11-22:
You will have to copy the assets from a Nucleus server. In doing so, you may find the [USD Path](https://docs.omniverse.nvidia.com/extensions/latest/ext_usd-paths.html) tool useful.