isaac-sim/IsaacLab #1309: Failed to query CUDA device count.

Jasmine-aiq, 2024-10-26:
when i run the isaac sim , it happened..How i fix it?
[image]

Jasmine-aiq, 2024-10-26:
[image]

jsmith-bdai, 2024-10-28:
Are you running Isaac Lab or bare Isaac Sim? What command are you running? What does `nvidia-smi` show?

In the future, please use the bug report and fill out the fields there so we can better help you:
[image]

RandomOakForest, 2024-11-22:
This may be resolved by rebooting or reinstalling the driver. Let us know if this issue persist. Thanks.

closed 2025-01-14 by RandomOakForest