isaac-sim/IsaacLab #1311: [Question] Error - ModuleNotFoundError: No module named 'omni.kit'

meghbhalerao, 2024-10-27:
When I try to run the following - 
`from omni.isaac.lab.envs import DirectRLEnv`

I get the following error - 

```
Traceback (most recent call last):
  File "/home/mb230/projects/human_ai_collab/human_ai_collab/environments/human_env.py", line 1, in <module>
    from omni.isaac.lab.envs import DirectRLEnv
  File "/home/mb230/projects/pkgs/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/envs/__init__.py", line 45, in <module>
    from . import mdp, ui
  File "/home/mb230/projects/pkgs/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/envs/mdp/__init__.py", line 18, in <module>
    from .actions import *  # noqa: F401, F403
  File "/home/mb230/projects/pkgs/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/envs/mdp/actions/__init__.py", line 8, in <module>
    from .actions_cfg import *
  File "/home/mb230/projects/pkgs/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/envs/mdp/actions/actions_cfg.py", line 9, in <module>
    from omni.isaac.lab.managers.action_manager import ActionTerm, ActionTermCfg
  File "/home/mb230/projects/pkgs/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/managers/__init__.py", line 13, in <module>
    from .action_manager import ActionManager, ActionTerm
  File "/home/mb230/projects/pkgs/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/managers/action_manager.py", line 18, in <module>
    import omni.kit.app
ModuleNotFoundError: No module named 'omni.kit'
```
My machine specfications are the following -
```
Operating System: Ubuntu 24.04.1 LTS              
 Kernel: Linux 6.8.0-45-generic
 Architecture: x86-64
```

I followed all the installation instructions here - https://isaac-sim.github.io/IsaacLab/main/source/setup/installation/pip_installation.html#

My conda environment is the following - 
```
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                       2_gnu    conda-forge
absl-py                   2.1.0                    pypi_0    pypi
antlr4-python3-runtime    4.9.3                    pypi_0    pypi
asttokens                 2.4.1                    pypi_0    pypi
boto3                     1.35.44                  pypi_0    pypi
botocore                  1.35.44                  pypi_0    pypi
bzip2                     1.0.8                h4bc722e_7    conda-forge
ca-certificates           2024.8.30            hbcca054_0    conda-forge
certifi                   2024.8.30                pypi_0    pypi
charset-normalizer        3.4.0                    pypi_0    pypi
click                     8.1.7                    pypi_0    pypi
cloudpickle               3.1.0                    pypi_0    pypi
contourpy                 1.3.0                    pypi_0    pypi
cycler                    0.12.1                   pypi_0    pypi
decorator                 4.4.2                    pypi_0    pypi
docker-pycreds            0.4.0                    pypi_0    pypi
egl-probe                 1.0.2                    pypi_0    pypi
exceptiongroup            1.2.2                    pypi_0    pypi
executing                 2.1.0                    pypi_0    pypi
farama-notifications      0.0.4                    pypi_0    pypi
filelock                  3.13.1                   pypi_0    pypi
fonttools                 4.54.1                   pypi_0    pypi
fsspec                    2024.2.0                 pypi_0    pypi
gitdb                     4.0.11                   pypi_0    pypi
gitpython                 3.1.43                   pypi_0    pypi
grpcio                    1.67.0                   pypi_0    pypi
gym                       0.23.1                   pypi_0    pypi
gym-notices               0.0.8                    pypi_0    pypi
gymnasium                 0.29.0                   pypi_0    pypi
h5py                      3.12.1                   pypi_0    pypi
hidapi                    0.14.0.post2             pypi_0    pypi
hydra-core                1.3.2                    pypi_0    pypi
idna                      3.10                     pypi_0    pypi
imageio                   2.36.0                   pypi_0    pypi
imageio-ffmpeg            0.5.1                    pypi_0    pypi
[... 113 lines cut ...]
tensorboard               2.18.0                   pypi_0    pypi
tensorboard-data-server   0.7.2                    pypi_0    pypi
tensorboardx              2.6.2.2                  pypi_0    pypi
tensordict                0.5.0                    pypi_0    pypi
termcolor                 2.5.0                    pypi_0    pypi
tk                        8.6.13          noxft_h4845f30_101    conda-forge
toml                      0.10.2                   pypi_0    pypi
torch                     2.4.0+cu121              pypi_0    pypi
torchvision               0.19.0                   pypi_0    pypi
tqdm                      4.66.5                   pypi_0    pypi
traitlets                 5.14.3                   pypi_0    pypi
trimesh                   4.5.0                    pypi_0    pypi
triton                    3.0.0                    pypi_0    pypi
typing-extensions         4.9.0                    pypi_0    pypi
tzdata                    2024.2                   pypi_0    pypi
urllib3                   2.2.3                    pypi_0    pypi
wandb                     0.12.21                  pypi_0    pypi
warp-lang                 1.4.1                    pypi_0    pypi
wcwidth                   0.2.13                   pypi_0    pypi
werkzeug                  3.0.4                    pypi_0    pypi
wheel                     0.44.0             pyhd8ed1ab_0    conda-forge
xz                        5.2.6                h166bdaf_0    conda-forge
```
Could someone help me out here? 
Thanks in advance!

larsrpe, 2024-10-28:
Did you launch the SimulationApp before importing? Most modules are not available before the app is lunched.

meghbhalerao, 2024-10-28:
Thanks. I modified my code in the following way (following here https://isaac-sim.github.io/IsaacLab/main/source/tutorials/00_sim/create_empty.html#launching-the-simulator) but I still get an error - below are details - 
Code Snippet - 
```
from __future__ import annotations

import math
import torch
from collections.abc import Sequence

from omni.isaac.lab.app import AppLauncher

from omni.isaac.lab_assets.cartpole import CARTPOLE_CFG

import omni.isaac.lab.sim as sim_utils
from omni.isaac.lab.assets import Articulation, ArticulationCfg
from omni.isaac.lab.envs import DirectRLEnv, DirectRLEnvCfg
```

and my error is - 
`ModuleNotFoundError: No module named 'omni.isaac'`

everything else is the same ... could you please help!

thanks!

meghbhalerao, 2024-10-28:
Found a fix - https://github.com/isaac-sim/IsaacLab/issues/257#issuecomment-2352011015 - just need to prepend the snippet mentioned in the linked issue to any "standalone" script that you want to run. Pasting the snippet (which needs to be prepended to run standalone scripts) here for added convenience - 

```
# Launch Isaac Sim Simulator first.
import argparse
from omni.isaac.lab.app import AppLauncher

# add argparse arguments
parser = argparse.ArgumentParser(description="My dummy script.")
# append AppLauncher cli args
AppLauncher.add_app_launcher_args(parser)
# parse the arguments
args_cli = parser.parse_args()

# launch omniverse app
app_launcher = AppLauncher(args_cli)
simulation_app = app_launcher.app
# Rest everything follows.
```

closed 2024-10-28 by meghbhalerao