isaac-sim/IsaacLab #1314: [Question] Type mismatch : expected 'GfQuatf', got 'GfQuatd' when I try to train robots

mitsu3291, 2024-10-28:
### Question

I use original robot model and made usd file.
I actually spawned the robot and saw the robot appears.
However, when I tried to train the robots, I came across the below error : 
`Traceback (most recent call last):
  File "/root/IsaacLab/source/standalone/workflows/rsl_rl/train.py", line 131, in <module>
    main()
  File "/root/IsaacLab/source/standalone/workflows/rsl_rl/train.py", line 86, in main
    env = gym.make(args_cli.task, cfg=env_cfg, render_mode="rgb_array" if args_cli.video else None)
  File "/isaac-sim/kit/python/lib/python3.10/site-packages/gymnasium/envs/registration.py", line 802, in make
    env = env_creator(**env_spec_kwargs)
  File "/root/IsaacLab/source/extensions/omni.isaac.lab_tasks/omni/isaac/lab_tasks/direct/my_robot/my_robot_simple_walk_env.py", line 88, in __init__
    super().__init__(cfg, render_mode, **kwargs)
  File "/root/IsaacLab/source/extensions/omni.isaac.lab_tasks/omni/isaac/lab_tasks/direct/my_robot/my_robot_base_env.py", line 95, in __init__
    super().__init__(cfg, render_mode, **kwargs)
  File "/root/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/envs/direct_rl_env.py", line 102, in __init__
    self._setup_scene()
  File "/root/IsaacLab/source/extensions/omni.isaac.lab_tasks/omni/isaac/lab_tasks/direct/my_robot/my_robot_base_env.py", line 115, in _setup_scene
    self.robot = Articulation(self.cfg.robot)
  File "/root/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/assets/articulation/articulation.py", line 97, in __init__
    super().__init__(cfg)
  File "/root/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/assets/rigid_object/rigid_object.py", line 58, in __init__
    super().__init__(cfg)
  File "/root/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/assets/asset_base.py", line 75, in __init__
    self.cfg.spawn.func(
  File "/root/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/sim/utils.py", line 275, in wrapper
    cloner.clone(prim_paths[0], prim_paths[1:], replicate_physics=False, copy_from_source=cfg.copy_from_source)
  File "/isaac-sim/exts/omni.isaac.cloner/omni/isaac/cloner/cloner.py", line 214, in clone
    xform_op_rot.Set(current_orientation)
pxr.Tf.ErrorException: 
	Error in 'pxrInternal_v0_22__pxrReserved__::UsdStage::_SetValueImpl' at line 6189 in file /builds/omniverse/usd-ci/USD/pxr/usd/usd/stage.cpp : 'Type mismatch for </World/envs/env_0/Robot.xformOp:orient>: expected 'GfQuatf', got 'GfQuatd''` 
	
Does anyone know this error?
I am using "direct" and copied the g1 reward settings to match the robot I am using.

jsmith-bdai, 2024-10-28:
We have seen this recently, but haven't root caused it. I also verified that I don't see this on 1.2, are you on an older version by chance?

Also, in the future, please use the bug report template if you're facing an error, thanks!
[image]

mitsu3291, 2024-10-30:
After the update, it started working! Thank you!

Understood about the bug report as well!

closed 2024-10-30 by mitsu3291