isaac-sim/IsaacLab #1318: [Bug Report] Incorrect Unit of length in CameraCfg

Autumnhxp, 2024-10-28:
If you are submitting a bug report, please fill in the following details and use the tag [bug].

### Describe the bug

The unit of focal length and horizontal aperture should be both in mm rather than cm to ensure the accuracy of the results in simulation.

### Steps to reproduce

Please try to provide a minimal example to reproduce the bug. Error messages and stack traces are also helpful.

### System Info

Describe the characteristic of your environment:

- Commit: [7b3ca4f]
- Isaac Sim Version: [4.1.0.0]
- OS: [Ubuntu 20.04]
- GPU: [RTX 4090]
- CUDA: [12.4]
- GPU Driver: [550.107.02]

### Additional context

The focal length of 35mm-50mm is close to the human perspective, 24cm (240mm) is telephoto lens. 

### Checklist

- [ x ] I have checked that there is no similar issue in the repo (**required**)
- [ x ] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

Add the criteria for which this task is considered **done**. If not known at issue creation time, you can add this once the issue is assigned.

- [ ] Criteria 1
- [ ] Criteria 2

jsmith-bdai, 2024-10-28:
cc: @pascal-roth, can you confirm this?

RandomOakForest, 2024-11-20:
Thanks for posting this. The default values for Isaac Lab are in cm and come from [Replicator](https://docs.omniverse.nvidia.com/py/replicator/1.11.16/source/extensions/omni.replicator.core/docs/API.html#cameras). I will forward this note to the Replicator team and close this for now.

closed 2024-11-20 by RandomOakForest