isaac-sim/IsaacLab #1319: [Question] H1 Arm Control

RiccardoZuppetti, 2024-10-28:
### Question

Hi,

Based on the current policy that allows to H1 to move in the surrouding environment, how would it be possible to control the movement of the humanoid's arms, while maintaining balance? Probably the policy in question does not take arm control into account.

jsmith-bdai, 2024-10-28:
That's correct, the current policy doesn't take arm control into account. 

You could create a new task and train it to control the arms, or modify the actions being sent to the arm's actuators from the trained policy (but this might not be robust)

RiccardoZuppetti, 2024-10-28:
> That's correct, the current policy doesn't take arm control into account.

Actually, I'm exploiting the same example built-in within Isaac Sim, this [one](https://docs.omniverse.nvidia.com/isaacsim/latest/isaac_lab_tutorials/tutorial_policy_deployment.html#unitree-h1-humanoid-example)

> You could create a new task and train it to control the arms, or modify the actions being sent to the arm's actuators from the trained policy (but this might not be robust)

How can I create a new task? Is there some reference to which I can refer? I've seen that for the arm control there are some tasks but are mostly for manipulators.

jsmith-bdai, 2024-10-29:
Yes, we have tutorials for creating tasks [here](https://isaac-sim.github.io/IsaacLab/main/source/tutorials/03_envs/create_manager_rl_env.html), although they are for simpler tasks. 

The H1 task was trained via https://github.com/isaac-sim/IsaacLab/blob/main/source/extensions/omni.isaac.lab_tasks/omni/isaac/lab_tasks/manager_based/locomotion/velocity/config/h1/flat_env_cfg.py I believe so that would be a good starting point to build off of

RiccardoZuppetti, 2024-11-13:
Once that a task has been trained, how can I exploit the generated policy in order to create my custom environment in IsaacSim? Is there something that could help me in this sense?

StafaH, 2026-09-26:
Closing this issue as it is stale. We have quite a few environments in core and contrib that can be used to learn how to build new tasks. Please use them as a starting off point for building further tasks. There are also examples now of sim2sim transfer to isaacsim that you can use for running policies.

closed 2026-09-26 by StafaH