isaac-sim/IsaacLab #1320: [Question] Documentation Questions regarding IsaacLab installation

meghbhalerao, 2024-10-28:
I am following the IsaacLab installation process here - https://isaac-sim.github.io/IsaacLab/main/source/setup/installation/pip_installation.html#installation-using-isaac-sim-pip, the first step about installing isaac sim using pip works fine and is clear.

However the second step about installing IsaacLab here - https://isaac-sim.github.io/IsaacLab/main/source/setup/installation/pip_installation.html#installing-isaac-lab is a little unclear to me (I may be missing something but please help me out). I have the following questions - 
1. We need to `cd IsaacLab` after cloning it, correct? This might be obvious, but I think it might be useful to make this clearer.
2. While installing IsaacLab, the conda environment should be active, correct? that is the conda env which we created in the previous step - https://isaac-sim.github.io/IsaacLab/main/source/setup/installation/pip_installation.html#installing-isaac-lab - I had this active and things seemed to work, that is the verifying installation script seemed to run etc, but wanted to clarify this!

Thanks for the help!

RandomOakForest, 2024-11-20:
Thanks for the suggestion. @jsmith-bdai please let us know if this has been resolved or if you need any help. Thanks.

AntoineRichard, 2026-08-27:
Thanks for taking the time to point these out, @meghbhalerao — ambiguous install steps are exactly the kind of thing worth reporting.

Both points have since been addressed by a docs reorganization: the old `pip_installation` page is gone, and the install guide now lives under `docs/source/setup/installation/` built around the uv-managed workflow. The `cd IsaacLab` step is now emitted explicitly by the clone snippets, and virtual-environment activation is spelled out step by step rather than assumed. Closing on that basis — please open a fresh docs issue if anything in the current guide is still unclear.

---
*🤖 This issue was reviewed as part of an AI-assisted triage of the Isaac Lab backlog, with automated verification against the current codebase. If we got this wrong, please reopen — we'd rather hear about it!*

closed 2026-08-27 by AntoineRichard