isaac-sim/IsaacLab #1324: [Question] Using convex decomposition and defining collisions masks in Isaac Lab

kbkartik, 2024-10-29:
Currently, I am migrating from Isaac Gym to Isaac Lab. 

In Isaac Gym, for convex decomposition, we used to set the following params as follows:

1. `asset.vhacd_enabled = True`
2. `asset.vhacd_params.resolution = 10000`

Similarly, for collisions, I used to set the bitmasks value to `filter` property in `isaacgym.gymapi.RigidShapeProperties` class.  

How can I achieve the same in Isaac Lab i.e., set convex decomposition parameters and define collision masks? I did not find relevant info for the same in the current documentation.

kbkartik, 2024-10-30:
Bumping this to get your help @Mayankm96 @kellyguo11

RandomOakForest, 2024-11-21:
Thanks for posting this. In Isaac Lab, `vhacd` is not used anymore, but there are convex decomposition attributes that can be modified through USD. See [this doc for reference](
https://docs.omniverse.nvidia.com/kit/docs/omni_usd_schema_physics/104.1/class_physx_schema_physx_convex_decomposition_collision_a_p_i.html).

closed 2025-01-14 by RandomOakForest