isaac-sim/IsaacLab #1325: [Question] How to model MuJoCo's friction loss parameter in Isaac Lab?

SuomiKP31, 2024-10-29:
### Question
I'm trying to reproduce some results from a MuJoCo codebase in Isaac Lab.
While PhysX has a friction parameter which can be set through the `ActuatorCfg` class in Isaac Lab, it is modelled differently compared to its MuJoCo counterpart.

MuJoCo's [friction loss](https://mujoco.readthedocs.io/en/stable/computation/index.html#friction-loss) parameter is a load-independent friction, which does not correlate to the normal force. It can be applied to revolute joints or sliders, and it has implicit unit corresponding to the joint type.

In PhysX, the friction parameter present on joints is a unitless coefficient, modelling Coulomb friction which scales with the normal force. Essentially the two simulators modelled the same thing using different approximation models.

Being a newcomer to Isaac Lab, my immediate thought is to calibrate my robot's friction coefficients in Isaac Sim so that it falls in similar fashion under gravity compared to MuJoCo. But I do wonder if there's a way to more quantitively model the friction, so that it is as close to MuJoCo as possible.

Is it possible to obtain the **combined force/torque applied to a joint after the actions have been applied to the actuators but before simulation update**, so that I can apply this static friction manually?

StephenWelch, 2024-10-30:
Also curious about this!

SuomiKP31, 2024-10-31:
I have previously used Omni-Isaac Gym Envs. I just looked back and found a function under ArticulationView called [get_measured_joint_forces](https://docs.omniverse.nvidia.com/py/isaacsim/source/extensions/omni.isaac.core/docs/index.html#omni.isaac.core.articulations.ArticulationView.get_measured_joint_forces) which seems to be what I need. However, Isaac Lab doesn't seem to be using this API. Can I use it in conjunction with the isaac.lab.assets.Articulation API?
I'm going to try this tomorrow to see if it leads to anywhere.

GiulioRomualdi, 2024-11-04:
Back in time I opened a similar question in the nvidia forum https://forums.developer.nvidia.com/t/how-to-implement-load-independent-static-friction-torque-in-isaacsim-revolute-joint/307662/1. Given the answer I think there is no a simple solution to implement dry friction, or static friction in isaaclab

StrainFlow, 2024-11-06:
It does not actually scale with the "normal force" but with the "transmitted joint force". Here is the documentation's description: 

`The joint friction is unitless and relates the magnitude of the spatial force [F_trans, T_trans] transmitted from parent to child link to the maximal friction force F_resist that may be applied by the solver to resist joint motion, per axis; i.e. |F_resist| <= coefficient * (|F_trans| + |T_trans|), where F_resist may refer to a linear force or torque depending on the joint axis.

The simulated friction effect is therefore similar to static and Coulomb friction. In order to simulate dynamic joint friction, use a joint drive with zero stiffness and zero velocity target, and an appropriately dimensioned damping parameter.` 

I'm digging into this. So far I have learned that if you are modeling joint friction, you should use the PGS solver, not the TGS solver.

RandomOakForest, 2024-11-22:
Thanks for posting this. Hopefully the discussion and suggested documentation helped. Will close this for now, but let us know may we be of further help. You may, for example, start a new post in our _Discussions_ section. Thanks.

closed 2024-11-22 by RandomOakForest