isaac-sim/IsaacLab #1329: [Question] Improving RL Training for Unitree G1: Adding Minimal Height Restriction for Center of Mass

XinyuSong123, 2024-10-29:
Hi, I recently conducted RL training with the Unitree G1 on flat terrain. The results show that, while it can move toward the commanded velocity, it struggles to stand up and exhibits a very unusual gait. After checking the reward settings, I found that there is no height restriction for the center of mass (presumably the torso link). Could you provide some guidance on how to add a minimum height restriction for the torso link? Thank you!

KyleM73, 2024-11-02:
Are you looking for a reward to track a desired center of mass height or a termination condition if the center of mass falls below a certain cutoff height?

XinyuSong123, 2024-11-06:
> Are you looking for a reward to track a desired center of mass height or a termination condition if the center of mass falls below a certain cutoff height?

Both reward options sound good to me. Additionally, could you provide guidance on implementing a reward that encourages the G1 to lift its legs at a specific frequency? Thank you!!!!

XinyuSong123, 2024-11-08:
Ah I've found the height settings

closed 2024-11-08 by XinyuSong123