isaac-sim/IsaacLab #1331: Bug Fixed for the fish eye config

wang59695487, 2024-10-29:
https://github.com/isaac-sim/IsaacLab/blob/9cc298e938c8903f4ba3d50a222c9c1f3a76c435/source/extensions/omni.isaac.lab/test/sim/test_spawn_sensors.py#L78

The issue originates from the project type in FisheyeCameraCfg(https://github.com/isaac-sim/IsaacLab/blob/main/source/extensions/omni.isaac.lab/omni/isaac/lab/sim/spawners/sensors/sensors_cfg.py#L171). To properly use the API in Isaac Sim Rep, it should be "fisheyePolynomial" rather than variations like "fisheye_polynomial".

   -  fisheye_orthographic -> fisheyeOrthographic
   -  fisheye_equidistant -> fisheyeEquidistant
   - fisheye_equisolid -> fisheyeEquisolid
   - fisheye_polynomial -> fisheyePolynomial
   - fisheye_spherical -> fisheyeSpherical

command-z-z, 2024-10-30:
You are right! I'm curious why such a big bug has not been fixed. I will raise a PR

command-z-z, 2024-10-31:
> https://github.com/isaac-sim/IsaacLab/blob/9cc298e938c8903f4ba3d50a222c9c1f3a76c435/source/extensions/omni.isaac.lab/test/sim/test_spawn_sensors.py#L78
>
> The issue originates from the project type in [...]

hi, I would like to know how you know the correct camera projection_type parameters. I checked the informations on [Replicator camera](https://docs.omniverse.nvidia.com/py/replicator/1.9.8/source/extensions/omni.replicator.core/docs/API.html#omni.replicator.core.create.camera) website. Only finding the following type, Acttually, `fisheye_polynomial` isn't the correct parameter.
[image].

command-z-z, 2024-10-31:
I found that the` projection_type` only supports the following six parameters.
[image]

PR #1361 (2024-12-07): Corrects fisheye camera projection types in spawner configuration
# Description

This MR fixes the support of fish-eye projection type. Actually, I don't find the proper fisheye type on [Replicator camera](https://docs.omniverse.nvidia.com/py/replicator/1.9.8/source/extensions/omni.replicator.core/docs/API.html#omni.replicator.core.create.camera) website (maybe it's out of style).

[image]

I only check the  `cameraProjectionType` in `Raw USD Properties` in Isaacsim GUI.

[image]

Fixes #1331

## Type of change

- Breaking change (fix or feature that would cause existing functionality to not w [... cut]
commit 5a3ade2b (2024-12-07): Corrects fisheye camera projection types in spawner configuration (#1361)

# Description

This MR fixes the support of fish-eye projection type. Actually, I don't
find the proper fisheye type on [Replicator
camera](https://docs.omniverse.nvidia.com/py/replicator/1.9.8/source/extensions/omni.replicator.core/docs/API.html#omni.replicator.core.create.camera)
website (maybe it's out of style).

![Screenshot from 2024-10-31
23-15-24](https://github.com/user-attachments/assets/7f826389-fb9e-474b-a54c-ae3c78e92d8b)

I only check the `cameraProjectionType` in `Raw USD Properties` in
Isaacsim GUI.

![S [... cut]
closed 2024-11-08 by RandomOakForest