isaac-sim/IsaacLab #1332: [Bug Report] dock_in function not working

junwang048bot, 2024-10-29:
If you are submitting a bug report, please fill in the following details and use the tag [bug].

### Describe the bug

Firstly I have created some viewports through code and assign each to a camera view, which works perfectly through code:
`window_1 = create_viewport_window("camera1")
window_1.viewport_api.set_active_camera("/World/envs/env_0/Robot/Camera1")
window_2 = create_viewport_window("camera2")
window_2.viewport_api.set_active_camera("/World/envs/env_0/Robot/Camera2")`

However, when I use the dock_in function following the IsaacSim code to achieve some layouts as shown in the picture below, I encounter issues. The dock_in function does not seem to be working.
Isaacsim Code:
`# Dock the second camera window
left_viewport = omni.ui.Workspace.get_window("Viewport")
right_viewport = omni.ui.Workspace.get_window("Viewport 2")
if right_viewport is not None and left_viewport is not None:
    right_viewport.dock_in(left_viewport, omni.ui.DockPosition.RIGHT)
`
[image]

My code: 
`# Dock the second camera window
window_2.dock_in(window_1, omni.ui.DockPosition.TOP, 0.5)
window_3.dock_in(window_2, .....)
......`

When I execute the code, nothing happens

### System Info

Describe the characteristic of your environment:

- Isaac Sim Version: 4.2.0
- OS: [e.g. Ubuntu 22.04]
- GPU: RTX 4090
- CUDA: 12.0
- GPU Driver: 550.107.02 
### Additional context

Add any other context about the problem here.

### Checklist

- [ ✔️] I have checked that there is no similar issue in the repo (**required**)
- [ ✔️] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

babakc, 2025-01-29:
I'm seeing this too. :(

VictorYudin, 2025-05-22:
Hello,

`dock_in()` must be executed only after the window objects have actually been created and registered with the UI workspace.  
Because window creation happens on the next render/update tick, the safest way is to:

1. Create the windows.  
2. Yield one frame (`await omni.kit.app.get_app().next_update_async()`).  
3. Call `dock_in()`.

Here is a minimal working example that you can drop in the Script Editor:

```python
import omni.ui as ui
import omni.kit.app
import asyncio

# Global storage so the windows survive for the lifetime of setup_viewports
windows = {}

async def setup_viewports():
    # Main viewport provided by Isaac Sim
    viewport = ui.Workspace.get_window("Viewport")

    # Create three additional windows
    windows["cam1"] = ui.Window("Camera 1", width=300, height=300)
    windows["cam2"] = ui.Window("Camera 2", width=300, height=300)
    windows["cam3"] = ui.Window("Camera 3", width=300, height=300)

    # Wait one frame so the UI system fully registers them
    await omni.kit.app.get_app().next_update_async()

    # Dock layout
    windows["cam1"].dock_in(viewport, ui.DockPosition.TOP)
    windows["cam2"].dock_in(viewport, ui.DockPosition.LEFT)

    # Wait one frame so the UI system fully docs "cam1"
    await omni.kit.app.get_app().next_update_async()
    windows["cam3"].dock_in(windows["cam1"], ui.DockPosition.LEFT)

# Schedule the coroutine
asyncio.ensure_future(setup_viewports())
```

Hope this helps!

animitta, 2025-09-14:
> 你好，
> 
> `dock_in()`必须在窗口对象实际创建并注册到 UI 工作区后执行。 由于窗口创建发生在下一个渲染/更新 tick，因此最安全的方法是：
> 
> 1. 创建窗口。
> 2. 产生一帧（`await omni.kit.app.get_app().next_update_async()`）。
> 3. 称呼`dock_in()`。
> 
> 这是一个可以放入脚本编辑器的最小工作示例：
> 
> import omni.ui as ui
> import omni.kit.app
> import asyncio
> 
> # Global storage so the windows survive for the lifetime of setup_viewports
> windows = {}
> 
> async def setup_viewports():
>     # Main viewport provided by Isaac Sim
>     viewport = ui.Workspace.get_window("Viewport")
> 
>     # Create three additional windows
>     windows["cam1"] = ui.Window("Camera 1", width=300, height=300)
>     windows["cam2"] = ui.Window("Camera 2", width=300, height=300)
>     windows["cam3"] = ui.Window("Camera 3", width=300, height=300)
> 
>     # Wait one frame so the UI system fully registers them
>     await omni.kit.app.get_app().next_update_async()
> 
>     # Dock layout
>     windows["cam1"].dock_in(viewport, ui.DockPosition.TOP)
>     windows["cam2"].dock_in(viewport, ui.DockPosition.LEFT)
> 
>     # Wait one frame so the UI system fully docs "cam1"
>     await omni.kit.app.get_app().next_update_async()
>     windows["cam3"].dock_in(windows["cam1"], ui.DockPosition.LEFT)
> 
> # Schedule the coroutine
> asyncio.ensure_future(setup_viewports())
> 希望这有帮助！

not working