isaac-sim/IsaacLab #1335: [Bug Report] Setting DirectRLEnv max_episode_length Causing Crash when Env is truncated

SuomiKP31, 2024-10-29:
### Describe the bug

When a DirectRLEnv is truncated by having a max_episode_length property set in gym.register, 

Isaac Lab will crash when a truncate happens. The root cause being a single "True" is returned from env.step(action) as "truncate", instead of a tensor.

### Steps to reproduce

When a DirectRLEnv is truncated by having a **max_episode_length** property set in gym.register, such as
```
gym.register(
    id="Isaac-Cartpole-Direct-v0",
    entry_point="omni.isaac.lab_tasks.direct.cartpole:CartpoleEnv",
    disable_env_checker=True,
    max_episode_steps=10,
    kwargs={
        "env_cfg_entry_point": CartpoleEnvCfg,
        "rl_games_cfg_entry_point": f"{agents.__name__}:rl_games_ppo_cfg.yaml",
        "rsl_rl_cfg_entry_point": f"{agents.__name__}.rsl_rl_ppo_cfg:CartpolePPORunnerCfg",
        "skrl_cfg_entry_point": f"{agents.__name__}:skrl_ppo_cfg.yaml",
        "sb3_cfg_entry_point": f"{agents.__name__}:sb3_ppo_cfg.yaml",
    },
)
```
Note that I set it to 10 to ensure a quick crash. Setting it for higher can delay the crash. If a truncate never happens, it doesn't crash.

running `.\isaaclab.bat -p .\source\standalone\workflows\sb3\train.py --task=Isaac-Cartpole-Direct-v0 --num_envs=64` will cause a crash with error message:

```
  File "C:\Users\mbran\miniconda3\envs\isaaclab\lib\site-packages\stable_baselines3\common\on_policy_algorithm.py", line 300, in learn
    continue_training = self.collect_rollouts(self.env, callback, self.rollout_buffer, n_rollout_steps=self.n_steps)
  File "C:\Users\mbran\miniconda3\envs\isaaclab\lib\site-packages\stable_baselines3\common\on_policy_algorithm.py", line 195, in collect_rollouts
    new_obs, rewards, dones, infos = env.step(clipped_actions)
  File "C:\Users\mbran\miniconda3\envs\isaaclab\lib\site-packages\stable_baselines3\common\vec_env\base_vec_env.py", line 206, in step
    return self.step_wait()
  File "C:\Work_shin\IsaacLab\source\extensions\omni.isaac.lab_tasks\omni\isaac\lab_tasks\utils\wrappers\sb3.py", line 239, in step_wait
    truncated = truncated.detach().cpu().numpy()
AttributeError: 'bool' object has no attribute 'detach'
```

[image]

Here, a single bool "True" is returned instead of a vector.
I also tested using rsl_rl, it crashes as well, just with a different error message.

### System Info

Describe the characteristic of your environment:

- Commit: e00d625 (Release 1.20)
- Isaac Sim Version: 4.2.0
- OS: Win 11
- GPU: RTX 4070 Ti S
- CUDA: 12.6
- GPU Driver: 566.03

### Additional context

If you want your episodes to end at the specific episode length, you can directly modify max_episode_s or code additional behaviors in _get_dones() in your environment. 

### Checklist

- [X] I have checked that there is no similar issue in the repo (**required**)
- [X] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

Add the criteria for which this task is considered **done**. If not known at issue creation time, you can add this once the issue is assigned.

- [ ] The Cartpole task does not crash when truncated with max_episode_steps set in gym.register
- [ ] Or, if this property in Environment Spec is ought to be ignored, output a warning saying setting this value does not have any effect, instead of crashing.

RandomOakForest, 2024-11-08:
Thanks for filing this. Will fix.

StafaH, 2026-09-20:
Fixing this would require excessive validation on the side of the wrappers. Such validation is not needed as it is expected that the gym registry should only be used to register the task and cfg entry points as shown in all the examples across isaaclab. The error should point the user to the correct fix naturally, so a validation in the wrappers is not needed.

closed 2026-09-20 by StafaH