isaac-sim/IsaacLab #1342: [Question] Missing References for Visual and Collision Attributes When Loading Local .usd Files

yilin404, 2024-10-30:
I appreciate your hard work on this project, and I hope to contribute by sharing an issue I've encountered.
**Description**: When using sim_utils.UsdFileCfg in Isaac Sim to import a .usd file, I encounter a warning message stating Missing references found for visual and collision attributes. However, I can successfully import the .usd file by dragging and dropping it, and it displays correctly.
My code is: 
```
CABINET_CFG = ArticulationCfg(
        prim_path="/World/envs/env_.*/Cabinet",
        spawn=sim_utils.UsdFileCfg(
            usd_path=f"{LOCAL_ISAAC_NUCLEUS_DIR}/Props/Sektion_Cabinet/sektion_cabinet_instanceable.usd",
            activate_contact_sensors=False,
        ),
        init_state=ArticulationCfg.InitialStateCfg(
            pos=(0.8, 0, 0.0),
            rot=(0.0, 0.0, 0.0, 1.0),
            joint_pos={
                "door_left_joint": 0.0,
                "door_right_joint": 0.0,
                "drawer_bottom_joint": 0.0,
                "drawer_top_joint": 0.0,
            },
        ),
        actuators={
            "drawers": ImplicitActuatorCfg(
                joint_names_expr=["drawer_top_joint", "drawer_bottom_joint"],
                effort_limit=87.0,
                velocity_limit=100.0,
                stiffness=10.0,
                damping=1.0,
            ),
            "doors": ImplicitActuatorCfg(
                joint_names_expr=["door_left_joint", "door_right_joint"],
                effort_limit=87.0,
                velocity_limit=100.0,
                stiffness=10.0,
                damping=2.5,
            ),
        },
    )
```
And the result is:
[image]
[image]
And when i dragg and dropp it into Isaac Sim, the result is:
[image]
**My Analysis**: I suspect this issue may be related to the relative path of the file, but I haven’t been able to resolve it. If you have any insights or suggestions, they would be greatly appreciated.

Thank you very much for your time and help!

AsawareeBhide, 2024-11-12:
Hi! Seems like something to do with how the USD path is set in ISAAC_NUCLEUS_DIR.
I imported this cabinet asset while running the [franka_cabinet_env.py](https://github.com/isaac-sim/IsaacLab/blob/main/source/extensions/omni.isaac.lab_tasks/omni/isaac/lab_tasks/direct/franka_cabinet/franka_cabinet_env.py) example and didn't observe the issue you're seeing.
Could you please confirm if the import statements in your code match those in the franka_cabinet_env example?

RandomOakForest, 2024-11-20:
We haven't been able to repro this issue. We'll close it for now, but please reopen if you still see this, or you have further information for us to review. Thanks.

closed 2024-11-20 by RandomOakForest