isaac-sim/IsaacLab #1343: [Proposal] Could it be possible to support pairwise contact forces?

KevinGeng3, 2024-10-30:
### Proposal

Could it be possible to support pairwise contact forces? In other words, could it be possible for us to get the contact forces between two objects/Actors?

### Motivation

When training dual-arm agents, I wish to penalize collision between arms. Therefore, I need to get the contact forces between them. Now I can only use contact sensor to get the forces that one arm have, but I have no way to find where the forces come from. This makes it difficult to determine whether a collision has occurred between the two arms.

### Alternatives

### Additional context

### Checklist

### Acceptance Criteria

zxhuang97, 2024-10-30:
I believe it's already supported. You can access the pairwise contact force using `sensor.data.force_matrix_w`. But be aware that 

> The reporting of the filtered contact forces is only possible as one-to-many. This means that only one sensor body in an environment can be filtered against multiple bodies in that environment. If you need to filter multiple sensor bodies against multiple bodies, you need to create separate sensors for each sensor body.

Code that might be helpful:
```
        self.scene.nut.spawn.activate_contact_sensors = True
        self.scene.contact_sensor = ContactSensorCfg(
            prim_path="{ENV_REGEX_NS}/Nut/factory_nut",
            filter_prim_paths_expr= ["{ENV_REGEX_NS}/Bolt/factory_bolt"],
            update_period=0.0,
        )
```
Also remember to set the `max_contact_data_count` in rigid_contact_view to appropriate number.

RandomOakForest, 2024-10-31:
Thank you for the proposal. This is not currently supported as you propose. As explained in [the current documentation](https://isaac-sim.github.io/IsaacLab/main/source/api/lab/omni.isaac.lab.sensors.html#contact-sensor) for `omni.isaac.lab.sensors` contact sensor, "you need to create a separate sensor for each [object] and filter it against the [other] object." We will tag this as a feature to consider on a future release.

KevinGeng3, 2024-10-31:
> I believe it's already supported. You can access the pairwise contact force using `sensor.data.force_matrix_w`. But be aware that
>
>
> Code that might be helpful:
>
> ``` [...]

Wow, thank you! It actually solved my problem.

KevinGeng3, 2024-10-31:
> Thank you for the proposal. This is not currently supported as you propose. As explained in [the current [...]

Thank you!

closed 2024-10-31 by KevinGeng3