isaac-sim/IsaacLab #1344: Issue on page /source/tutorials/03_envs/create_manager_rl_env.html

FerryRain, 2024-10-30:
I ruined the script "run_cartpole_rl_env.py" and report an error called
"Creating window for environment.
2024-10-30 13:31:44 [13,417ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-10-30 13:31:44 [13,417ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-10-30 13:31:44 [13,417ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-10-30 13:31:44 [13,417ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-10-30 13:31:44 [13,417ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-10-30 13:31:44 [13,417ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-10-30 13:31:44 [13,417ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-10-30 13:31:44 [13,418ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-10-30 13:31:44 [13,418ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-10-30 13:31:44 [13,418ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-10-30 13:31:44 [13,418ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-10-30 13:31:44 [13,418ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-10-30 13:31:44 [13,418ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Core' for removal
Traceback (most recent call last):
  File "/home/ferry/Code/Research/IsaacLab_Learn/RL_Policy/Create_a_Manager_based_RL_Env/run_cartpole_rl_env.py", line 24, in <module>
    env = ManagerBasedRLEnv(cfg=env_cfg,debug_vis=False)
  File "/home/ferry/Code/Sim/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/envs/manager_based_rl_env.py", line 75, in __init__
    super().__init__(cfg=cfg)
  File "/home/ferry/Code/Sim/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/envs/manager_based_env.py", line 148, in __init__
    self._window = self.cfg.ui_window_class_type(self, window_name="IsaacLab")
  File "/home/ferry/Code/Sim/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/envs/ui/manager_based_rl_env_window.py", line 37, in __init__
    self._create_debug_vis_ui_element("commands", self.env.command_manager)
  File "/home/ferry/Code/Sim/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/envs/ui/base_env_window.py", line 385, in _create_debug_vis_ui_element
    checked=elem.cfg.debug_vis,
AttributeError: 'NoneType' object has no attribute 'debug_vis'
"

Mayankm96, 2024-10-30:
We request you to file a bug report using the appropriate template for the same. It is hard for us to debug based on the description above.

FerryRain, 2024-10-30:
> We request you to file a bug report using the appropriate template for the same. It is hard for us to debug based on the description above.

Where can I find the template?

RandomOakForest, 2024-10-30:
Please use the "New Issue" button at the top. For the row of "Bug Report" click on "Get Started" and then fill in the form. Add a title and replace the form text with your details. You may leave the sections subtitles (###) in the text of the form while replacing the text that follows. The form starts with:
```
If you are submitting a bug report, please fill in the following details and use the tag [bug].

### Describe the bug

A clear and concise description of what the bug is.

### Steps to reproduce

Please try to provide...
```
Thank you.

FerryRain, 2024-10-31:
> Please use the "New Issue" button at the top. For the row of "Bug Report" click on "Get Started" and then fill in the form. Add a title and replace the form text with your details. You may leave the [...]

Thanks, I have already submitted a new Bug Report.

closed 2024-10-31 by FerryRain