isaac-sim/IsaacLab #1350: [Question] I encountered a coordinate transformation issue while setting up the controller

amatsukaze233, 2024-10-31:
### Question

When I tried to modify the template program Using a task-space controller and added a translation and rotation to my robot's InitialStateCfg, the rotation did not seem to take effect. 
Additionally,the marker needs to include root_pose_w[:, 0:3] to display correctly.
`goal_marker.visualize(ik_commands[:, 0:3] + scene.env_origins+root_pose_w[:, 0:3], ik_commands[:, 3:7])`

`robot = HSR_CFG.replace(prim_path="{ENV_REGEX_NS}/HSR1",
                             init_state=ArticulationCfg.InitialStateCfg(pos=(1,7.5,0),rot=(0.707,0,0,-0.707),
                                joint_pos={"carriage": 6.0,"joint1": 0.0,"joint2": 0.0,"joint3": 0.0,"joint4": 0.0,
                                           "joint5": 1.57,"joint6": 0.0,"handtoleft": 0.0,"handtoright": 0.0,},
                                joint_vel={"carriage":0.1})
                            )`

Translation only
[image]
Translation and rotation
[image]

RandomOakForest, 2024-11-08:
This may be a bug. If you still have this issue, could you please file a new issue as a Bug Report with steps to reproduce and related info? Thanks!

amatsukaze233, 2024-11-18:
> This may be a bug. If you still have this issue, could you please file a new issue as a Bug Report with steps to reproduce and related info? Thanks!

Thank you for your reply!I have already submitted a bug report at https://github.com/isaac-sim/IsaacLab/issues/1433

closed 2024-11-18 by amatsukaze233