isaac-sim/IsaacLab #1351: [Bug Report] The DOC https://isaac-sim.github.io/IsaacLab/main/source/tutorials/03_envs/create_manager_rl_env.html report an error

FerryRain, 2024-10-31:
If you are submitting a bug report, please fill in the following details and use the tag [bug].

### Describe the bug

I met an error when I ran the file in the tutorials of the doc files " https://isaac-sim.github.io/IsaacLab/main/source/tutorials/03_envs/create_manager_rl_env.html".

### Steps to reproduce

`python -p source/standalone/tutorials/03_envs/run_cartpole_rl_env.py --num_envs 32`

```
[INFO][AppLauncher]: Loading experience file: /home/ferry/Code/Sim/IsaacLab/source/apps/isaaclab.python.kit
Loading user config located at: '/home/ferry/.local/share/ov/pkg/isaac-sim-4.2.0/kit/data/Kit/Isaac-Sim/4.2/user.config.json'
[Info] [carb] Logging to file: /home/ferry/.local/share/ov/pkg/isaac-sim-4.2.0/kit/logs/Kit/Isaac-Sim/4.2/kit_20241030_232947.log
2024-10-30 15:29:47 [0ms] [Warning] [omni.kit.app.plugin] No crash reporter present, dumps uploading isn't available.
2024-10-30 15:29:47 [20ms] [Warning] [omni.ext.plugin] [ext: __pycache__] Extensions config 'extension.toml' doesn't exist '/home/ferry/Code/Research/IsaacLab_Learn/RL_Policy/Create_a_Manager_based_RL_Env/__pycache__' or '/home/ferry/Code/Research/IsaacLab_Learn/RL_Policy/Create_a_Manager_based_RL_Env/__pycache__/config'

|---------------------------------------------------------------------------------------------|
| Driver Version: 560.35.03     | Graphics API: Vulkan
|=============================================================================================|
| GPU | Name                             | Active | LDA | GPU Memory | Vendor-ID | LUID       |
|     |                                  |        |     |            | Device-ID | UUID       |
|     |                                  |        |     |            | Bus-ID    |            |
|---------------------------------------------------------------------------------------------|
| 0   | NVIDIA GeForce RTX 4090 D        | Yes: 0 |     | 24810   MB | 10de      | 0          |
|     |                                  |        |     |            | 2685      | 0f159d70.. |
|     |                                  |        |     |            | 1         |            |
|=============================================================================================|
| OS: 20.04.6 LTS (Focal Fossa) ubuntu, Version: 20.04.6, Kernel: 5.15.0-107-generic
| XServer Vendor: The X.Org Foundation, XServer Version: 12013000 (1.20.13.0)
| Processor: 13th Gen Intel(R) Core(TM) i7-13700 | Cores: 16 | Logical: 24
|---------------------------------------------------------------------------------------------|
| Total Memory (MB): 31779 | Free Memory: 15439
| Total Page/Swap (MB): 2047 | Free Page/Swap: 1994
|---------------------------------------------------------------------------------------------|
2024-10-30 15:29:52 [4,678ms] [Error] [omni.kit.stage_templates.new_stage] Failed to create directory /home/ferry/Documents/Kit/apps/Isaac-Sim/scripts/new_stage
2024-10-30 15:29:53 [5,694ms] [Error] [omni.kit.menu.edit.scripts.screenshot_page] Failed to create directory /home/ferry/Documents/Kit/shared/screenshots/
2024-10-30 15:29:53 [5,810ms] [Warning] [omni.replicator.core.scripts.annotators] Annotator PostProcessDispatch is already registered, overwriting annotator template
2024-10-30 15:29:53 [6,013ms] [Warning] [omni.kit.widget.cache_indicator.cache_state_menu] Unable to detect Omniverse Cache Server. Consider installing it for better IO performance.
2024-10-30 15:29:54 [6,680ms] [Error] [rtx.neuraylib.plugin] Cannot find `Documents/Kit/shared` directory.
2024-10-30 15:29:57 [9,833ms] [Warning] [omni.isaac.lab.envs.manager_based_env] Seed not set for the environment. The environment creation may not be deterministic.
[INFO]: Base environment:
	Environment device    : cuda:0
	Environment seed      : None
	Physics step-size     : 0.008333333333333333
	Rendering step-size   : 0.016666666666666666
	Environment step-size : 0.016666666666666666
[INFO]: Time taken for scene creation : 2.235946 seconds
[INFO]: Scene manager:  <class InteractiveScene>
	Number of environments: 32
	Environment spacing   : 4.0
	Source prim name      : /World/envs/env_0
	Global prim paths     : []
	Replicate physics     : True
[INFO]: Starting the simulation. This may take a few seconds. Please wait...
2024-10-30 15:30:00 [13,133ms] [Warning] [omni.hydra.scene_delegate.plugin] Calling getBypassRenderSkelMeshProcessing for prim /World/envs/env_0/Robot/pole/visuals.proto_mesh_0_id0 that has not been populated
[INFO]: Time taken for simulation start : 1.058464 seconds
[INFO] Command Manager:  <CommandManager> contains 0 active terms.
+------------------------+
|  Active Command Terms  |
+--------+-------+-------+
| Index  | Name  |  Type |
+--------+-------+-------+
+--------+-------+-------+

[INFO] Action Manager:  <ActionManager> contains 1 active terms.
+----------------------------------+
|  Active Action Terms (shape: 1)  |
+-------+--------------+-----------+
| Index | Name         | Dimension |
[... 83 lines cut ...]
```

For more information on this, check: https://www.markdownguide.org/extended-syntax/#fenced-code-blocks

-->

### System Info

Describe the characteristic of your environment:

- Commit: 7d40df3
- Isaac Sim Version: 4.2.0
- OS: Ubuntu 20.04
- GPU:RTX 4090 D
- CUDA: 12.6
- GPU Driver: 560.35.03

### Additional context

Add any other context about the problem here.

### Checklist

- I have checked that there is no similar issue in the repo (**required**)
- [ ] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

FlimFlamm, 2024-10-31:
I'm also encountering this error. It's a bit confusing... on line 385 of `base_env_window.py`, when `elem,.cfg.debug_vis` is accessed, the NoneType error is thrown. Strangely, when I tried to debug print it, it both printed the value, and then threw the NoneType error for the debug print.... Here's a full stack trace after running: `HYDRA_FULL_ERROR=1 ./isaaclab.sh -p source/standalone/workflows/sb3/train.py --task Isaac-Cartpole-v0 --num_envs 64`

```py
# omitted above prints that seemed irrelevant)
Creating window for environment.
Error executing job with overrides: []
Traceback (most recent call last):
  File "/home/chris/Documents/isaaclab99/IsaacLab/source/standalone/workflows/sb3/train.py", line 156, in <module>
    main()
  File "/home/chris/Documents/isaaclab99/IsaacLab/source/extensions/omni.isaac.lab_tasks/omni/isaac/lab_tasks/utils/hydra.py", line 102, in wrapper
    hydra_main()
  File "/home/chris/.local/lib/python3.10/site-packages/hydra/main.py", line 94, in decorated_main
    _run_hydra(
  File "/home/chris/.local/lib/python3.10/site-packages/hydra/_internal/utils.py", line 394, in _run_hydra
    _run_app(
  File "/home/chris/.local/lib/python3.10/site-packages/hydra/_internal/utils.py", line 457, in _run_app
    run_and_report(
  File "/home/chris/.local/lib/python3.10/site-packages/hydra/_internal/utils.py", line 223, in run_and_report
    raise ex
  File "/home/chris/.local/lib/python3.10/site-packages/hydra/_internal/utils.py", line 220, in run_and_report
    return func()
  File "/home/chris/.local/lib/python3.10/site-packages/hydra/_internal/utils.py", line 458, in <lambda>
    lambda: hydra.run(
  File "/home/chris/.local/lib/python3.10/site-packages/hydra/_internal/hydra.py", line 132, in run
    _ = ret.return_value
  File "/home/chris/.local/lib/python3.10/site-packages/hydra/core/utils.py", line 260, in return_value
    raise self._return_value
  File "/home/chris/.local/lib/python3.10/site-packages/hydra/core/utils.py", line 186, in run_job
    ret.return_value = task_function(task_cfg)
  File "/home/chris/Documents/isaaclab99/IsaacLab/source/extensions/omni.isaac.lab_tasks/omni/isaac/lab_tasks/utils/hydra.py", line 99, in hydra_main
    func(env_cfg, agent_cfg, *args, **kwargs)
  File "/home/chris/Documents/isaaclab99/IsaacLab/source/standalone/workflows/sb3/train.py", line 106, in main
    env = gym.make(args_cli.task, cfg=env_cfg, render_mode="rgb_array" if args_cli.video else None)
  File "/home/chris/anaconda3/envs/isaaclab99/lib/python3.10/site-packages/gymnasium/envs/registration.py", line 802, in make
    env = env_creator(**env_spec_kwargs)
  File "/home/chris/Documents/isaaclab99/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/envs/manager_based_rl_env.py", line 75, in __init__
    super().__init__(cfg=cfg)
  File "/home/chris/Documents/isaaclab99/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/envs/manager_based_env.py", line 148, in __init__
    self._window = self.cfg.ui_window_class_type(self, window_name="IsaacLab")
  File "/home/chris/Documents/isaaclab99/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/envs/ui/manager_based_rl_env_window.py", line 37, in __init__
    self._create_debug_vis_ui_element("commands", self.env.command_manager)
  File "/home/chris/Documents/isaaclab99/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/envs/ui/base_env_window.py", line 385, in _create_debug_vis_ui_element
    checked=elem.cfg.debug_vis,
AttributeError: 'NoneType' object has no attribute 'debug_vis'```

If I set line 385 to True or False instead of checking the debuig_vis parameter, then it the environment works fine and the cartpoles learn.

### System Info

Describe the characteristic of your environment:
    Isaac Sim Version: 4.2.0
    OS: Ubuntu 22.10 (Kinetic)
    GPU:RTX 4090
    CUDA: 12.2
    GPU Driver: 535.129.03

### Additional Context:

I manually set up my environment rather than using docker, so possibly I missed a step and am missing something or have the wrong version of something. Let me know if there is any other information I can snag to help diagnose the problem.

FlimFlamm, 2024-10-31:
Apologies for omitting the commit. I was on `002fec...` (several commits ago)

Updating did indeed fix the problem for me. Thanks so much!

FerryRain, 2024-10-31:
> I don't see the commit linked above: 7d40df3
> 
> This should have been fixed on the latest main some time ago. Can you please update your fork and check?

Thank you very much. I solved the problem when I checked the latest commit.

closed 2024-10-31 by FerryRain