isaac-sim/IsaacLab #1362: [Bug Report] Final observation in step

EladSharony, 2024-11-02:
### Describe the bug

In the [`step`](https://github.com/isaac-sim/IsaacLab/blob/f7350c72b3e7ac310eda00cbc325b8f80f31e66a/source/extensions/omni.isaac.lab/omni/isaac/lab/envs/direct_rl_env.py#L283) method, the `next_obs` is overwritten by the `reset_obs`.

Assume we have reached a terminal/truncated state, i.e., `self.reset_buf.any() == True`. Currently, these are the steps performed in `step`:

> 1. Pre-process the actions before stepping through the physics.
> 2. Apply the actions to the simulator and step through the physics in a decimated manner.
> 3. Compute the reward and done signals.
> **-> If we call `self._get_observations()` here, we will get the actual `next_obs`.**
> 4. Reset environments that have terminated or reached the maximum episode length.
> **-> When we call `self._get_observations()` here, we get the `reset_obs`.**
> 5. Apply interval events if they are enabled.
> 6. Compute observations. **Here is where `self._get_observations()` is actually called.**

Following the same logic as in [Gymnasium v0.29.0](https://github.com/Farama-Foundation/Gymnasium/blob/v0.29.0/gymnasium/vector/sync_vector_env.py), I suggest adding a `final_observation` key in the `info` dictionary:

``` python
# ... 
# -- reset envs that terminated/timed-out and log the episode information
reset_env_ids = self.reset_buf.nonzero(as_tuple=False).squeeze(-1)
if len(reset_env_ids) > 0:
    for key in self.obs_buf:
        # store the final observation for terminated/timed-out environments
        self.extras.update({"final_observation": {key: self.obs_buf[key][reset_env_ids]}})

    self._reset_idx(reset_env_ids)
    # if sensors are added to the scene, make sure we render to reflect changes in reset
    if self.sim.has_rtx_sensors() and self.cfg.rerender_on_reset:
        self.sim.render()
elif "final_observation" in self.extras:
    # remove the final observation if no envs have terminated/timed-out
    self.extras.pop("final_observation")
# ... 
```

And in the training loop, we can then handle it as follows:

``` python 
obs, infos = env.reset()

obs = torch.as_tensor(obs['policy'])
for _ in range(100):
    action = torch.as_tensor(env.action_space.sample())
    next_obs, rewards, terminations, truncations, infos = env.step(action)

    next_obs = next_obs['policy']
    real_next_obs = next_obs.clone()

    if truncations.any():
        truncations_env_ids = truncations.nonzero(as_tuple=False).squeeze(-1)
        real_next_obs[truncations_env_ids] = infos["final_observation"]['policy']

    # (s_t, a_t, r_t, s_{t+1}, d)
    transition = {'observation': obs, 'action': action, 'reward': rewards,
                  'next_observation': real_next_obs, 'dones': terminations}

    obs = next_obs
```

**Some notes:**
- Perhaps there is a cleaner way to remove the `final_observation` key when returning `self.extras` - which is a class instance (why is that actually?).
- In the new [Gymnasium v1.0.0](https://github.com/Farama-Foundation/Gymnasium/releases/tag/v1.0.0), they propose a different solution, which introduces a different problem—the number of samples added to the replay buffer isn't constant.
- I'm not sure if the same issue occurs in `ManagerBasedEnv` (I personally don't use it).

### Steps to reproduce

Run any `DirectRLEnv` environment, add breakpoints before and after `self._reset_idx(reset_env_ids)`, and call `self._get_observations()`.

### System Info

Describe the characteristic of your environment:

- Commit: a56291d0
- Isaac Sim Version: 4.2.0
- OS: Ubuntu 24.04.1
- GPU: RTX4090
- CUDA: 12.6
- GPU Driver: 560.35.03

### Additional context
-

### Checklist

- [x] I have checked that there is no similar issue in the repo (**required**)
- [x] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

Add the criteria for which this task is considered **done**. If not known at issue creation time, you can add this once the issue is assigned.
- [ ]

RandomOakForest, 2024-11-04:
Thanks for filing this. This is not a bug in itself, it is currently the mode of operation of most libraries. These algorithms do not need to know what the final observation was. We could implement this in the future, though. I will tag it as an enhancement ("Proposal") for the team. Thank you.

EladSharony, 2024-11-04:
Hi @RandomOakForest, I appreciate the prompt response.

I understand that this might be the current behavior in the library, but I'm not sure I am familiar with any algorithm that doesn't need the **correct** transition. 

Currently, when an environment resets, the transition provided is `(s_T, a_T, r_T, s_0_hat, d)`, where `s_0_hat` is the initial state of a new episode rather than the true final state `s_{T+1}` of the current episode. This means the transition does not represent a valid sequence in the environment, as the next state does not logically follow from the previous state and action.

Additionally, there is currently no straightforward way to recover the true final state using a wrapper without re-implementing the entire `step` method, which adds unnecessary complexity.

Regarding other libraries, again, I'm not aware of any that consider this the correct operation.

RandomOakForest, 2024-11-08:
Thank you bringing this to our attention. Your comment has been duly noted.

AntoineRichard, 2026-08-27:
Thanks for the write-up, @EladSharony — terminal observations are now supported natively. Set `compute_final_obs=True` on the environment config and the pre-reset observation is published under `extras["final_obs"]` in `DirectRLEnv` and `ManagerBasedRLEnv` — and per agent under `extras[<agent>]["final_obs"]` in `DirectMARLEnv`. It's opt-in, so existing behavior is unchanged. This landed in https://github.com/isaac-sim/IsaacLab/pull/6219 (`source/isaaclab/isaaclab/envs/direct_rl_env.py`, `manager_based_rl_env.py`).

Closing as resolved — please reopen if the exposed buffer doesn't cover your use case.

---
*🤖 This issue was reviewed as part of an AI-assisted triage of the Isaac Lab backlog, with automated verification against the current codebase. If we got this wrong, please reopen — we'd rather hear about it!*

closed 2026-08-27 by AntoineRichard