isaac-sim/IsaacLab #1363: [Proposal] Adding project options for docker compose command

ZiwenZhuang, 2024-11-03:
### Proposal

Adding project name option for `docker/container.py` script to generate different container and image name.

### Motivation

We have different people using docker on a shared bare-metal cluster. We would like to make the container separate for each person that are sharing the same machine.

Also, installing learning frame using docker script installs them all. I would like to add options to it by setting `docker/.env.base` file.

### Checklist

- [x] I have checked that there is no similar issue in the repo (**required**)
- [x]  I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

Add the criteria for which this task is considered **done**. If not known at issue creation time, you can add this once the issue is assigned.

- [ ] Adding command line option in `docker/container.py` script.
- [ ] Adding interface from `docker/.env.base` to `docker/Dockerfile.base` so that the RL framework installed in the container can be chosen.

RandomOakForest, 2024-11-04:
Thanks for filing this. Will tag for follow up as an enhancement.

Mayankm96, 2025-09-09:
We have added support for Docker suffix names: https://github.com/isaac-sim/IsaacLab/pull/2172

Please re-open this issue if the above does not solve the problem. Thank you!

closed 2025-09-09 by Mayankm96