isaac-sim/IsaacLab #1364: Support Needed: Implementing RL Policy for Omni-directional Ridgeback_Kuka Mobile Manipulator in Isaac Sim

sinaqahremani, 2024-11-04:
I am developing a reinforcement learning (RL) policy for a mobile manipulator composed of a Clearpath Ridgeback omni-directional base and a Kuka arm. The goal is to control the robot to reach specified positions and orientations accurately. The robot setup includes two LiDAR sensors (mounted on the front and back) and an IMU sensor for enhanced navigation.

Here is the current status and issues I'm encountering:

**1. Omni-directional Mobility Issue:** The URDF file of the robot has been successfully imported into Isaac Sim and converted to a USD file, but the Ridgeback base does not perform as an omni-directional robot. I would appreciate guidance on configuring it to move as intended.

**2. Sensor Setup:** I need to set up LiDAR and IMU sensors within Isaac Sim. Any advice on configuring these sensors for accurate data collection and compatibility with the RL framework would be valuable.

**3. Using IsaacLab for RL Implementation:** My final goal is to use IsaacLab for both simulating and implementing the RL environment. Guidance on best practices, tool integrations, and steps to set up the RL environment with IsaacLab would be greatly appreciated.

Thank you in advance for any insights or resources to resolve these challenges!

RandomOakForest, 2024-11-05:
Thanks for posting this, we may have to move it into Discussions. Regarding 1, we could use more details on your URDF/USD setup. Please try to narrow down the issue and create a new issue as a Question or Bug Report if necessary.

Regarding sensors, [available sensors in Isaac Lab can be found here](https://isaac-sim.github.io/IsaacLab/main/source/api/lab/omni.isaac.lab.sensors.html#module-omni.isaac.lab.sensors), there are also other modalities (LiDAR, etc.), in Isaac Sim that may be useful. 

Regarding RL, there are [tutorials](https://isaac-sim.github.io/IsaacLab/main/source/tutorials/index.html) and [how-to guides](https://isaac-sim.github.io/IsaacLab/main/source/how-to/index.html) in the Isaac Lab's documentation explaining the tasking framework, how to set up a new environment, comparisons of RL libraries, and how to set up your own RL library. This should be a good place to start.

sinaqahremani, 2024-11-05:
Thank you for your response. 
Is there any resource/tutorial on how to make a 4 wheel omnidirectional mobile robot? 
Regarding sensors, I am not sure if I need to implement them in IsaacSim / USD file or just using IsaacLab.

RandomOakForest, 2024-11-08:
We are aiming to have new tutorials up soon. We'll keep you posted. The tutorials currently in the docs of Isaac Lab and Isaac Sim are the only ones we can provide for now.

AntoineRichard, 2026-08-27:
Thanks for laying out your Ridgeback-Kuka project, @sinaqahremani, and sorry for the slow reply. This spans URDF import, LiDAR/IMU setup and general RL guidance rather than a specific Isaac Lab defect, so there's nothing concrete for us to track. The docs (https://isaac-sim.github.io/IsaacLab/release/3.0.0/) have grown a lot since 2024, and Discussions or the NVIDIA developer forums are usually faster for robot-model questions.

Closing for inactivity — please reopen, or open a focused issue, if you hit a concrete bug.

---
*🤖 This issue was reviewed as part of an AI-assisted triage of the Isaac Lab backlog, with automated verification against the current codebase. If we got this wrong, please reopen — we'd rather hear about it!*

closed 2026-08-27 by AntoineRichard