isaac-sim/IsaacLab #1365: Simulating Grading of Granular Terrain with Grader Vehicle

sinaqahremani, 2024-11-04:
I am interested in simulating interactions with granular materials, specifically rocks and stones of various sizes, to replicate scenarios where a grader vehicle levels a terrain filled with granular material.

I would like to know:

1. **Feasibility**: Is it possible to simulate granular materials, such as rocks and stones, in the environment? If so, what are the limitations or considerations to keep in mind?

2. **Grader Interaction with Granular Terrain**: Can the grader vehicle effectively interact with and manipulate this granular material within the simulation, accurately reflecting real-world grading tasks?

3. **Steps and Tools**: What would be the recommended tools, plugins, or specific settings to achieve realistic interactions between the grader and granular terrain?

Thank you for any guidance on setting up this type of simulation!

StrainFlow, 2024-11-05:
Something like this could be modeled using WARP: https://github.com/NVIDIA/warp

sinaqahremani, 2024-11-05:
@StrainFlow You think it is not possible to do that using IsaacLab?

StrainFlow, 2024-11-13:
I think it should be possible, I haven't done any warp integration myself yet.

RandomOakForest, 2025-02-11:
Will close this for now, but feel free to open a new _discussion_ may we be of further help. Thanks.

closed 2025-02-11 by RandomOakForest