isaac-sim/IsaacLab #1373: [Question] How to use marker in manager-based environment?

BRoln7, 2024-11-05:
### Question

I want to visualize paths by `visualization_markers` in manager-based environment, but now manager-based environment seems doesn't support `omni.isaac.lab.markers`, so how can I do it?

StrainFlow, 2024-11-15:
Can you share how you did this?

BRoln7, 2024-11-16:
`from omni.isaac.lab.markers import VisualizationMarkersCfg, VisualizationMarkers`, then don't define markers in the SceneCfg, define them after importing the modules.

AntoineRichard, 2026-08-27:
Thanks for asking, @BRoln7 — and for coming back to post your own solution. Markers aren't declared as scene entities in a manager-based `SceneCfg`; you instantiate them directly, e.g. `markers = VisualizationMarkers(VisualizationMarkersCfg(prim_path=..., markers={...}))`, and call `markers.visualize(...)` with your poses. `scripts/demos/markers.py` is a complete working example. The how-to guide walks through it end to end: https://isaac-sim.github.io/IsaacLab/release/3.0.0/source/how-to/draw_markers.html

Closing as answered — please reopen if any part of the marker workflow is still unclear.

---
*🤖 This issue was reviewed as part of an AI-assisted triage of the Isaac Lab backlog, with automated verification against the current codebase. If we got this wrong, please reopen — we'd rather hear about it!*

closed 2026-08-27 by AntoineRichard