isaac-sim/IsaacLab #1374: Prime not showing in semantic_segmentation

jacobtho314, 2024-11-05:
### Question

Currently, my camera’s semantic segmentation, when read as a numpy array, only shows the floor as ID 0, and the sky as ID 255. All other objects/prims in the scene are invisible, as if they don’t exist. How do I give each prim a unique ID for the semantic_segmentation data type of the camera? I would like to be able to determine how many unique objects are in a scene (camera’s POV only).

jacobtho314, 2024-11-05:
This is helpful, and my camera is now doing instance segmentation, but I couldn't find any documentation/information anywhere on adding instance IDs to prims. I see I can do semantic types, but I'm interested in tracking each unique object in a scene

closed 2024-11-05 by jacobtho314

RandomOakForest, 2024-11-08:
You will need the [instance ID segmentation annotator](https://docs.omniverse.nvidia.com/extensions/latest/ext_replicator/annotators_details.html#instance-id-segmentation).

jacobtho314, 2024-11-09:
This isn’t necessary. You can find the ID to class labels from the camera data’s info dict! This is why I asked this question multiple times, because, truthfully, nobody seems to know how Isaac Lab actually works.