isaac-sim/IsaacLab #1376: [Proposal] Sample more diverse rigid body material when randomizing

lukasfro, 2024-11-05:
### Proposal

I noticed that the number of **unique** sampled property tuples for the materials is determined by the argument `num_buckets`. From the name, I would have assumed that each of the three properties is divided into `num_buckets` buckets and the actual material is then sampled independently from each bucket, so that the maximum number of different materials would be `num_buckets ** 3`. 

### Motivation

Sampling the material properties as proposed above would increase the domain randomization in a relatively memory-efficient way. Moreover, given that the values for `num_buckets` in the examples are relatively small (64 and 250), some users might have fewer randomized materials as they expect. 

### Alternatives

Alternatively, one could of course simply increase the value of `num_buckets` to some large value. However, this would unnecessarily increase the size/memory of the `self.material_buckets`  attribute. 

### Additional context

Recently, a bug was fixed for the randomization of the rigid body materials #1140. I do not know whether the proposed behavior was originally intended or not.

### Checklist

- [x] I have checked that there is no similar issue in the repo (**required**)

### Acceptance Criteria

TODO: I'm happy to open a PR and add criteria in case the proposal is accepted.

RandomOakForest, 2024-11-05:
Thank you for posting this. We will review.

RandomOakForest, 2025-02-14:
Following up, if you'd like to submit a PR, we would review it. Thanks.