isaac-sim/IsaacLab #1377: [Bug Report] Very small typo in comment

meghbhalerao, 2024-11-05:
https://github.com/isaac-sim/IsaacLab/blob/main/source/extensions/omni.isaac.lab/omni/isaac/lab/sim/schemas/schemas_cfg.py#L32

I think the comment here should mention velocity instead of position.

Thanks!

RandomOakForest, 2024-11-05:
Thanks for catching this copy-paste mistake. We will fix it.

AntoineRichard, 2026-08-27:
Nice catch, @meghbhalerao. The `solver_velocity_iteration_count` docstring that incorrectly read "Solver position iteration counts" was fixed in https://github.com/isaac-sim/IsaacLab/pull/2112, which lists this issue directly. The schema configs have since moved to `source/isaaclab_physx/isaaclab_physx/sim/schemas/schemas_cfg.py`, where every occurrence now correctly says "Solver velocity iteration counts".

Closing as fixed — please reopen or send a PR if you spot the stale wording elsewhere in the tree.

---
*🤖 This issue was reviewed as part of an AI-assisted triage of the Isaac Lab backlog, with automated verification against the current codebase. If we got this wrong, please reopen — we'd rather hear about it!*

closed 2026-08-27 by AntoineRichard