isaac-sim/IsaacLab #1386: [Question] --writable issues

alextong1010, 2024-11-06:
### Question
`Running this command: ./docker/cluster/cluster_interface.sh job --task Isaac-Velocity-Rough-Anymal-C-v0 --headless --video --enable_cameras`
in a SLURM cluster left me with this issue in the error log:

```
WARNING: nv files may not be bound with --writable
WARNING: By using --writable, Singularity can't create /n destination automatically without overlay or underlay
FATAL:   container creation failed: mount /var/lib/singularity/mnt/session/n->/n error: while mounting /var/lib/singularity/mnt/session/n: destination /n doesn't exist in container
```

I tried removing the --writable flag, which caused additional issues with accessing or writing to directories (/isaac-sim).
i.e. `[omni.kit.stage_templates.new_stage] Failed to create directory /isaac-sim/kit/data/documents/Kit/apps/Isaac-Sim/scripts/new_stage
2024-11-06 19:44:04 [9,435ms] [Error] [omni.kit.app.plugin] failed to open file '/isaac-sim/kit/data/Kit/Isaac-Sim/4.2/user.config.json' - unable to save the dictionary
2024-11-06 19:44:05 [9,997ms] [Error] [rtx.neuraylib.plugin] Cannot find Documents/Kit/shared directory.`

kellyguo11, 2024-11-10:
@pascal-roth would you know something about this?

pascal-roth, 2024-11-11:
The '--writable' flag is necessary. I will try to reproduce this error on my side. To do so, could you please tell me which docker and singularity version you used to create the singularity image.

alextong1010, 2024-11-11:
Hi @pascal-roth, 
These are the versions.
Docker version: 27.3.1
Singularity version: 1.3.4
Apptainer version: 1.3.4

alextong1010, 2024-11-22:
Hi @pascal-roth, just curious, were you able to reproduce the error?

RandomOakForest, 2025-01-14:
Thanks for following up. This has to do with a failed mounted volume in a docker run command or similar. Do you still face this issue in the latest release? Thanks

RandomOakForest, 2025-02-11:
Will close this for now. Please try Isaac Lab 2.0 and let us know if we can be of further help. Thanks.

closed 2025-02-11 by RandomOakForest