isaac-sim/IsaacLab #1387: [Question] Group/mimic actuator

xhsonny, 2024-11-06:
Hi, I may miss something obvious here but I could not find related info.

I am trying to work with `mimic` joints. Based on the tutorial [here](https://isaac-sim.github.io/IsaacLab/v1.2.0/source/overview/core-concepts/actuators.html#actuator-groups), there is "Mimic Group". But I am not able to find any actuator group that can do this. 

I saw there was `GripperActuatorGroupCfg` but it was removed quickly.

What is current way to implement mimic group? Thanks!

AsawareeBhide, 2024-11-13:
Hi! The [Adding Mimic Joint section](https://docs.omniverse.nvidia.com/isaacsim/latest/advanced_tutorials/tutorial_advanced_rigging_complex_structures.html#adding-mimic-joint) might help to get started.

xhsonny, 2024-11-15:
Thanks. Close this issue for now.

closed 2024-11-15 by xhsonny