isaac-sim/IsaacLab #1388: How to add the semantic ID to the semantic tags

jacobtho314, 2024-11-07:
### Question
 
I saw a maintainer post the following dictionary for semantic tags and their corresponding colors:

{
    "(0, 0, 0, 0)": {
        "class": "BACKGROUND"
    },
    "(0, 0, 0, 255)": {
        "class": "UNLABELLED"
    },
    "(140, 255, 25, 255)": {
        "class": "cube"
    },
    "(140, 25, 255, 255)": {
        "class": "cylinder"
    },
    "(255, 197, 25, 255)": {
        "class": "cone"
    }
}

is it possible to programmatically generate this with semantic IDs? Where can I find documentation on how to do this with Isaac Lab?

RandomOakForest, 2024-11-07:
It is. You have to use the Replicator extension and [follow this example](https://docs.omniverse.nvidia.com/extensions/latest/ext_replicator/semantics_schema_editor.html#programmatically-defining-semantics). This takes care of assigning the class labels or semantic tags. Then, using the [semantic segmentation annotator](https://docs.omniverse.nvidia.com/extensions/latest/ext_replicator/annotators_details.html#semantic-segmentation) you will get a json file with a mapping for the labels.

jacobtho314, 2024-11-07:
This is incredibly frustrating; I basically need to re-write my sim from scratch using Replicator

RandomOakForest, 2024-11-16:
You may try a similar approach to the one suggested in [this discussion](https://github.com/isaac-sim/IsaacLab/discussions/1110). Alternatively, Replicator can be imported as any other extension and used as mentioned above.

RandomOakForest, 2024-11-22:
Hopefully the suggested documentation and discussion helped. Will close this for now, but let us know may you need further help. Thanks.

closed 2024-11-22 by RandomOakForest