isaac-sim/IsaacLab #1389: [Question] How to Configure PhysxCfg in omni.isaac.lab.sim API for Isaac Lab?

H-Hisamichi, 2024-11-07:
Hello everyone,

I am working on reinforcement learning with a custom hexapod, but I'm encountering a "collisionStackSize buffer overflow" error. I'd like to expand the buffer size to fix this. However, I'm not sure where and how to set `gpu_collision_stack_size` in my RL code (possibly in `def _setup_scene`).

If anyone has any pointers, I'd really appreciate it.

Thank you!

AsawareeBhide, 2024-11-07:
Hi! Could you please try setting it in your config class as shown [here](https://github.com/isaac-sim/IsaacLab/blob/main/source/extensions/omni.isaac.lab_tasks/omni/isaac/lab_tasks/direct/allegro_hand/allegro_hand_env_cfg.py#L38)?

H-Hisamichi, 2024-11-08:
Hello @AsawareeBhide,

Thank you for the advice.
I was able to allocate the buffer successfully.

H-Hisamichi, 2024-11-08:
Hello @AsawareeBhide,

A new error has come up:
```[Error] [omni.physx.plugin] PhysX error: Patch buffer overflow detected, please increase its size to at least 167788 in the scene desc!```

RL training runs fine at first, but after a while, this error appears. In this case, which buffer in `PhysxCfg` should I increase?

H-Hisamichi, 2024-11-10:
Increasing the `gpu_max_rigid_patch_count` buffer resolved the error.
Thank you for your help!

closed 2024-11-10 by H-Hisamichi