isaac-sim/IsaacLab #1390: [Bug Report] ContactSensor data is all zero in CPU mode

SuomiKP31, 2024-11-07:
### Describe the bug

sensor.data is not actively updated when a task is run with --device cpu flag.

### Steps to reproduce

For the quickest repro, modify anymal_c_env.py -> AnimalCEnv, to print out ``` self._contact_sensor.last_air_time ``` or ``` net_forces_w ```.

Then, run
```
.\isaaclab.bat -p .\source\standalone\workflows\rsl_rl\train.py --task=Isaac-Velocity-Flat-Anymal-C-Direct-v0 --device cpu --num_envs=8
```
All the values we get are zero, even if the feet sensors are clearly making contact with the ground.

Remove the --device cpu flag and run again. In GPU mode we get non-zero values.

### System Info

Describe the characteristic of your environment:

- Commit: e00d625 (Release 1.20)
- Isaac Sim Version: 4.2.0
- OS: Win 11
- GPU: RTX 4070 Ti S
- CUDA: 12.6
- GPU Driver: 566.03

### Additional context

This is probably not anything urgent as we would want to run robotics stuff on the GPU anyway. I was just super confused and thought I did something wrong - the document should at least state that the Contact Sensor does not work in CPU mode.

I tested using different workflows, one is using SB3 (my own env), and the above example is using rsl_rl. 

I have double checked that my sensor's update function is correctly executing.

If you bump into this issue when trying to use Contact Sensor, also checkout [this note](https://github.com/isaac-sim/IsaacLab/issues/571). It's very helpful!

### Checklist

- [X] I have checked that there is no similar issue in the repo (**required**)
- [X] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

Add the criteria for which this task is considered **done**. If not known at issue creation time, you can add this once the issue is assigned.

- [ ] Contact Sensor correctly returns data collected in CPU mode.

RandomOakForest, 2024-11-07:
Thank you for filling this. Will tag for follow up by the team.

shafeef901, 2024-11-18:
I can confirm that this issue persists. [This line](https://github.com/isaac-sim/IsaacLab/blob/b3ecfe3d3cd494ad6a43951d579f7b1735d3ec41/source/extensions/omni.isaac.lab/omni/isaac/lab/sensors/contact_sensor/contact_sensor.py#L323) which gets the net contact forces from `contact_physx_view` returns all zeroes.

Mayankm96, 2025-03-25:
Should be fixed via: https://github.com/isaac-sim/IsaacLab/pull/1861

closed 2025-03-25 by Mayankm96