isaac-sim/IsaacLab #1394: Trouble getting Camera data

InDroKalena, 2024-11-07:
### Isaac Sim Version 
2023.1.1

### Operating System
Ubuntu 22.04

### GPU Information 
NVIDIA CUDA 12.2 Driver Version 535.183.01

### Description
I am working on a simulation using Isaac Sim, ROS2, and (hopefully) Zed cameras. When I upload my Zed USD and initialize it as a Camera, everything runs smoothly and works as expected in the GUI. However, when I try to print .get_rgb() and .get_depth() to obtain data, I get [] and None, respectively.

### Steps to Reproduce

In a class called ControllerRobot:

    def setup_zed_camera(self):
        zed_prim_path = "/World/ZED_X"
        stage = omni.usd.get_context().get_stage()

        prims_utils.create_prim(
            prim_path="/World/ZED_X",
            prim_type="Xform",
            usd_path="(my_path_to)/ZED_X.usd"
        )

        left_camera_prim = stage.GetPrimAtPath(f"{zed_prim_path}/base_link/ZED_X/CameraLeft")
        right_camera_prim = stage.GetPrimAtPath(f"{zed_prim_path}/base_link/ZED_X/CameraRight")
        
        if not left_camera_prim.IsValid() or not right_camera_prim.IsValid():
            raise Exception("Could not find CameraLeft or CameraRight prims within the ZED_X model")
        
        orientation = euler_to_quaternion(0, 0, 0)
        orientation_array = np.array([orientation.GetReal(), orientation.GetImaginary()[0], orientation.GetImaginary()[1], orientation.GetImaginary()[2]])
        
        self.zed_camera_left = Camera(
            prim_path=str(left_camera_prim.GetPath()),
            name="zed_camera_left",
            position=np.array([0, 0, 1]),  
            orientation=orientation_array,
            frequency=20,
            resolution=(1280, 720)
        )
        self.zed_camera_left.initialize()
        self.zed_camera_left.add_distance_to_image_plane_to_frame()
        
        self.zed_camera_right = Camera(
            prim_path=str(right_camera_prim.GetPath()),
            name="zed_camera_right",
            position=np.array([0, 0, 1]), 
            orientation=orientation_array,
            frequency=20,
            resolution=(1280, 720)
        )
        self.zed_camera_right.initialize()
        self.zed_camera_right.add_distance_to_image_plane_to_frame()

    def get_zed_data(self):
        if self.zed_camera_left is not None and self.zed_camera_right is not None:
            try:
                left_rgb = self.zed_camera_left.get_rgb()
                left_depth = self.zed_camera_left.get_depth()
                right_rgb = self.zed_camera_right.get_rgb()
                right_depth = self.zed_camera_right.get_depth()
                return left_rgb, left_depth, right_rgb, right_depth
            except Exception as e:
                print(e)
        return None, None, None, None

And then in main:

def main(args=None):
    controller_robot = ControllerRobot()
    time.sleep(5)
    try:
        while simulation_app.is_running():
            simulation_app.update()
            controller_robot.update(1/60)
            rgb_data, depth_data, left_data, right_data = controller_robot.get_zed_data()
            print(rgb_data)
            print(depth_data)
            print(left_data)
            print(right_data)
etc.

### Error Messages
No error messages, everything runs smoothly! I do get this warning though: WARNING: Annotator 'distance_to_image_plane' contains no data. Returning None

## Additional Information
### What I've Tried
I've tried using simpler cameras and printing their data, but I get [] and None from them as well. I've tried the suggestion from the forum below but also nothing.

### Related Issues
https://forums.developer.nvidia.com/t/empty-ndarray-is-returned-from-camera-get-rgb-on-isaacsim-api/292731/2

### Additional Context
I haven't been able to get collisions working with my robot and the background in the 2023.1.1 version, so I'm wondering if the cameras just aren't "seeing" any obstacles?

RandomOakForest, 2024-11-08:
It seems you need to add to each of your `Camera` calls the annotators list:
```
data_types=["rgb", "distance_to_image_plane"],
```
See, for example, [this tutorial](https://isaac-sim.github.io/IsaacLab/main/source/tutorials/04_sensors/add_sensors_on_robot.html#the-code).

RandomOakForest, 2024-11-15:
Hi, if you still have this issue or need further help, let us know. Will close this issue for now, feel free to reopen it. Thanks for posting this.

closed 2024-11-15 by RandomOakForest