isaac-sim/IsaacLab #1395: [Question] Error when using RayCaster with multiple environments

SourabhPrasad, 2024-11-07:
Hi,
I'm working on setting up a DirectRL environment that includes a robot, a height scanner, and a building mesh. My goal is to be able to spawn multiple instances of this environment. 

As there are multiple instances, I have set the `mesh_prim_paths` attribute of `RayCasterCfg()` to `mesh_prim_paths=["/World/envs/env_.*/MeshObj"]`. This results in an `AttributeError: 'RayCaster' object has no attribute 'drift'` due to `ValueError: Prim at path '/World/envs/env_0/MeshObj' is not valid.` 

The program works correctly if `num_envs = 1`  and `mesh_prim_paths=["/World/envs/env_0/MeshObj"]`

Can multiple height scanners be spawned where the `mesh_prim_path` uses the environment namespace? Or could the error be due to the height scanner being initialised before the mesh?

#### Errors
```
Traceback (most recent call last):
  File "/home/sourabh/project/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/sensors/sensor_base.py", line 64, in <lambda>
    lambda event, obj=weakref.proxy(self): obj._initialize_callback(event),
  File "/home/sourabh/project/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/sensors/sensor_base.py", line 271, in _initialize_callback
    self._initialize_impl()
  File "/home/sourabh/project/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/sensors/ray_caster/ray_caster.py", line 153, in _initialize_impl
    self._initialize_warp_meshes()
  File "/home/sourabh/project/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/sensors/ray_caster/ray_caster.py", line 172, in _initialize_warp_meshes
    mesh_prim = sim_utils.get_first_matching_child_prim(
  File "/home/sourabh/project/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/sim/utils.py", line 570, in get_first_matching_child_prim
    raise ValueError(f"Prim at path '{prim_path}' is not valid.")
ValueError: Prim at path '/World/envs/env_.*/MeshObj' is not valid.
```

#### RayCasterCfg()
```python
height_scanner_cfg = RayCasterCfg(
        prim_path="/World/envs/env_.*/robot/base",
        offset=RayCasterCfg.OffsetCfg(pos=(1.0, 0.0, 2.0)),
        attach_yaw_only=True,
        pattern_cfg=patterns.GridPatternCfg(resolution=0.1, size=[2.5, 2.5]),
        debug_vis=True,
        mesh_prim_paths=["/World/envs/env_.*/MeshObj"],
```

#### _setup_scene()
```python
self.mesh_obj = sim_utils.spawn_from_usd(
            "/World/envs/env_.*/MeshObj",
            self.cfg.mesh_obj,
            (0.0, 0.0, 3.0),
            math_utils.quat_from_euler_xyz(
                torch.tensor(0),
                torch.tensor(0),
                torch.tensor(0),
            )
        )

        # add robots
        self.cfg.robot_cfg.spawn.num_envs = self.num_envs
        self._robot = Articulation(self.cfg.robot_cfg)
        self.scene.articulations["robot"] = self._robot

        # add sensors
        self._height_scanner = RayCaster(self.cfg.height_scanner_cfg)
        self.scene.sensors["height_scanner"] = self._height_scanner

        # add terrain
        self.cfg.terrain.num_envs = self.num_envs
        self.cfg.terrain.env_spacing = self.scene.cfg.env_spacing
        self.cfg.terrain.seed = self.seed
        self._terrain = self.cfg.terrain.class_type(self.cfg.terrain)

        # clone, filter, and replicate
        self.scene.filter_collisions(global_prim_paths=[self.cfg.terrain.prim_path])

        # add lights
        light_cfg = sim_utils.DomeLightCfg(intensity=2000.0, color=(0.75, 0.75, 0.75))
        light_cfg.func("/World/Light", light_cfg)
```

RandomOakForest, 2024-11-15:
Hi, let us know if you need further help. Will close this issue for now. Thanks for posting this.

closed 2024-11-15 by RandomOakForest

sunsleaf, 2024-12-23:
May I ask if you have further plan to fixing this issue, i.e., supporting regex and multiple meshes for raycaster?