isaac-sim/IsaacLab #1397: [Question] How to make an omnidirectional wheeled mobile robot

sinaqahremani, 2024-11-08:
### Question

I am working on a mobile manipulator robot from Clearpath company. The mobile base is a Ridgeback robot, which is an omnidirectional robot, and the arm is a Kuka arm. I have the URDF file of the robot and I am able to convert it to a USD file. However, the base doesn't move like an omnidirectional robot. I need tips on how to make it like an omnidirectional robot. 

Here is the urdf file of the Ridgeback robot:
https://github.com/ridgeback/ridgeback/tree/melodic-devel/ridgeback_description/urdf

I appreciate your assistance and your time.

RandomOakForest, 2024-11-15:
It's taken more time than we thought to sort this out. This will eventually be part of our docs. Thanks for posting this. We will aim to have a suggested solution next week.

shaoxiang, 2024-12-23:
I found that the official Clearpath RIDGEBACK_FRANKA model has a problem. 
The chassis has no wheels. 
Is there a problem with the omni-directional Mecanum in the simulation? 
Can omni-directional robots be trained in Isaac Lab?
[image]

RandomOakForest, 2025-01-02:
Thank you for posting this, @shaoxiang. However, this looks like a separate problem than the original topic above. Please submit a new question if you still have this issue. Thank you.

closed 2025-01-14 by RandomOakForest