isaac-sim/IsaacLab #1399: [Bug Report] Not able to use LiveStream and Streaming CLient

wzds2015, 2024-11-08:
If you are submitting a bug report, please fill in the following details and use the tag [bug].

### Describe the bug

All my installation went well. I've tested several headless simulations, and they look good. I finally tested livestream mode and streaming client. My GUI is blank. Any idea?

### Steps to reproduce
./isaaclab.sh -p source/standalone/demos/quadrupeds.py --livestream 1 (then open Streaming Client on Macbook to connect the server)

output from sudo ufw status
```
47995:48012/tcp            ALLOW       Anywhere                  
47995:48012/udp            ALLOW       Anywhere                  
49000:49007/tcp            ALLOW       Anywhere                  
49000:49007/udp            ALLOW       Anywhere                  
49100/tcp                  ALLOW       Anywhere                  
49100/udp                  ALLOW       Anywhere                  
5900/tcp                   ALLOW       Anywhere                  
5900/udp                   ALLOW       Anywhere                  
8211/tcp                   ALLOW       Anywhere                  
8211/udp                   ALLOW       Anywhere                  
22/tcp                     ALLOW       Anywhere                  
47995:48012/tcp (v6)       ALLOW       Anywhere (v6)             
47995:48012/udp (v6)       ALLOW       Anywhere (v6)             
49000:49007/tcp (v6)       ALLOW       Anywhere (v6)             
49000:49007/udp (v6)       ALLOW       Anywhere (v6)             
49100/tcp (v6)             ALLOW       Anywhere (v6)             
49100/udp (v6)             ALLOW       Anywhere (v6)             
5900/tcp (v6)              ALLOW       Anywhere (v6)             
5900/udp (v6)              ALLOW       Anywhere (v6)             
8211/tcp (v6)              ALLOW       Anywhere (v6)             
8211/udp (v6)              ALLOW       Anywhere (v6)             
22/tcp (v6)                ALLOW       Anywhere (v6)  
```

Screenshot showing GUI is empty
[image]

### System Info

Describe the characteristic of your environment:

- Commit: note this is IssacLab commit: b9a49caebc912b86f2ace0709c08d9884d167cda
- Isaac Sim Version: IssacLab 1.2.0
- OS: Ubuntu 22.04
- GPU: RTX A6000 50GB
- CUDA: 12.4
- GPU Driver: 550.120
- vulkan: 1.13.204
- Local machine: Macbook M2
- Streaming Client: MacOS version 1.3.4

### Additional context

Add any other context about the problem here.

### Checklist

- [ ] I have checked that there is no similar issue in the repo (**required**)
- [ ] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

Add the criteria for which this task is considered **done**. If not known at issue creation time, you can add this once the issue is assigned.

- [ ] Criteria 1
- [ ] Criteria 2

wzds2015, 2024-11-12:
pop up this question

RandomOakForest, 2024-11-13:
Thanks for submitting this. We are reviewing this issue with our engineering team and will post back here soon.

vrodriguezf, 2024-11-14:
same here, in my case with livestream 2 (webRTC). I can see the play button, but after that nothing gets streamed. I can see the simulation log on the terminal though

wzds2015, 2024-11-14:
> same here, in my case with livestream 2 (webRTC). I can see the play button, but after that nothing gets streamed. I can see the simulation log on the terminal though

Yes. I also tried WebRTC and have the same problem as yours.

francescotaioli, 2024-12-05:
I wanted to visualize sim results in a **remote headless ssh machine.**

I had some difficulties in doing that, but I managed to solve the issue. 
I'll report the information and provide some more detailed step-by-step guide.

**System Info**
```
Ubuntu 22.04.4 LTS
isaacsim  4.2.0.2
omni-isaac-lab   0.27.16
omni-isaac-lab_assets   0.1.4 
omni-isaac-lab_tasks      0.10.13
Cuda 12.1
```
**How to**
- Install ```Isaac Lab``` from https://isaac-sim.github.io/IsaacLab/main/source/setup/installation/pip_installation.html
- Verify the installation with ```./isaaclab.sh -p source/standalone/tutorials/00_sim/create_empty.py```. 

You shouldn't encounter an error, but you also won't see a window because we're working on a headless remote SSH machine without a windowing system.

**How to visualize the sim?**
- Install Nvidia Omniverse from  https://developer.nvidia.com/omniverse#section-getting-started
- Install and launch the **Streaming client** from  https://docs.omniverse.nvidia.com/streaming-client/latest/user-manual.html. This should open the Omniverse Streaming Client window. In the ```server``` input text box, insert your ```remote ssh machine IP address```. Do NOT click ```connect``` yet.
- In the remote machine, launch a sim. For example:
``` ./isaaclab.sh -p source/standalone/workflows/sb3/train.py --task Isaac-Cartpole-v0 --num_envs 1 --headless --livestream 1```
--livestream option is for live stream, with ```1``` for omniverse streaming client and ```2``` for webRTC.
- you should now see in the remote machine ```[INFO]: Starting the simulation. This may take a few seconds. Please wait...``` and some logging information. 
- Now, on the local machine, in the ```omniverse streaming client``` click to ```connect```

vrodriguezf, 2024-12-05:
Thank you for the update. That's still not applicable to me because my local machine does not have a GPU and therefore I cannot install the official streaming client.

francescotaioli, 2024-12-06:
@vrodriguezf, are you sure such a requirement exists? I don't have an NVIDIA GPU on my local PC either (I have a Intel Iris Xe Graphics)

vrodriguezf, 2024-12-06:
Oh I remember reading that in the documentation, but I might be wrong. I'm gonna try and report, thank you!

Yonggie, 2025-01-08:
Hi thank you for the detailed tips.
I also tried to use such approach but the IsaacSim seems to stuck in some bug and could not render the full scene I designed.

using Omniverse Streaming Client 103.1 and Isaacsim 4.2.0.
- If I use both ``headless`` and ``livestream1``, it'll show ``ModuleNotFoundError: No module named 'omni.kit.usd'``.
- If i only use ``headless``, the client could not connect to the server.

I used ``compatible`` settings provided by the tutorial app ``isaaclab.backwards.compatible.kit``, command being ``python source/standalone/tutorials/00_sim/create_empty.py  --headless --experience=/IsaacLab/source/apps/isaaclab.backwards.compatible.kit``, it'll barely show the service and the terminal keeps showing error
```
Traceback (most recent call last):
  File "/isaac-sim/kit/exts/omni.kit.viewport.ready/omni/kit/viewport/ready/viewport_ready.py", line 282, in __on_event
    waiting_for = self.__viewport_ready.viewport_handle
AttributeError: 'NoneType' object has no attribute 'viewport_handle'
...
Traceback (most recent call last):
  File "/isaac-sim/kit/exts/omni.kit.viewport.ready/omni/kit/viewport/ready/viewport_ready.py", line 282, in __on_event
    waiting_for = self.__viewport_ready.viewport_handle
AttributeError: 'NoneType' object has no attribute 'viewport_handle'
```

the service looks like this, and I could not move my view:
[image]

Yonggie, 2025-01-10:
> Hi thank you for the detailed tips. I also tried to use such approach but the client could not connect to the service. using Omniverse Streaming Client 103.1 and Isaacsim 4.2.0.
> 
> I complete container installation based on [documentation](https://docs.omniverse.nvidia.com/isaacsim/latest/installation/install_workstation.html), and start the service by `./runheadless.native.sh`, but it failed to connect to the service. I've also tried to add `livestream 1` and `headless`, same case.
> 
> I have confirmed that required ports are exposed to the host server:
> 
> ```
> CONTAINER ID   IMAGE                            COMMAND                  CREATED        STATUS                      PORTS
> 
>     NAMES
> d0ad681b9044   nvcr.io/nvidia/isaac-sim:4.2.0   "bash"                   23 hours ago   Up 23 hours                 8011-8012/tcp, 8211/tcp, 8891/tcp, 8899/tcp, 47995-48012/tcp, 47995-48012/udp, 49000-49007/tcp, 49100/tcp, 49000-49007/udp, 0.0.0.0:113->22/tcp, [::]:113->22/tcp   isaac-sim
> ```
> 
> ![图片](https://private-user-images.githubusercontent.com/31330044/401454607-69c0e021-9643-45e6-bc61-23025b655253.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzY0MTM4NjYsIm5iZiI6MTczNjQxMzU2NiwicGF0aCI6Ii8zMTMzMDA0NC80MDE0NTQ2MDctNjljMGUwMjEtOTY0My00NWU2LWJjNjEtMjMwMjViNjU1MjUzLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTAxMDklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwMTA5VDA5MDYwNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTA5ZjU1YTEzYjEwMjc2NTU3MmI1NWFlYjRlNzc2NDM0ZjUyNTgwZWY5ZmMwOWNmMGM4ZDkyMzNmNGQwN2Y5MWMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.Mv2BhfSszr7Lg453Ia4uX74pD8-yhnqH8GC8G76kgKE)

After several trials, The conncection could **ONLY** be built via docker run in ``host network mode``...

RandomOakForest, 2025-01-14:
Thank you for posting this. We will bring this to the attention of our Isaac Sim team.

RandomOakForest, 2025-02-11:
Closing this issue for now. Please reopen it if you need further help while trying the latest Isaac Lab and Isaac Sim. Thank you.

closed 2025-02-11 by RandomOakForest