isaac-sim/IsaacLab #1400: [Bug Report] Physx error on contact sensor

supersglzc, 2024-11-09:
Hi, 

In my task, I created 10 contact sensors to detect the collision between robot fingers and objects. However, after training around 60M steps, the simulation always crashed with the following error:
```
2024-11-09 19:44:40 [14,677,396ms] [Error] [omni.physx.plugin] Cuda context manager error, simulation will be stopped and new cuda context manager will be created.

2024-11-09 19:44:40 [14,677,397ms] [Error] [omni.physx.plugin] PhysX error: GPU compressContactStage1 fail to launch kernel stage 1!!, FILE /builds/omniverse/physics/physx/source/gpunarrowphase/src/PxgNarrowphaseCore.cpp, LINE 9566

2024-11-09 19:44:40 [14,677,397ms] [Error] [omni.physx.plugin] Cuda context manager error, simulation will be stopped and new cuda context manager will be created.

2024-11-09 19:44:40 [14,677,397ms] [Error] [omni.physx.plugin] PhysX error: GPU compressContactStage2 fail to launch kernel stage 1!!, FILE /builds/omniverse/physics/physx/source/gpunarrowphase/src/PxgNarrowphaseCore.cpp, LINE 9607

2024-11-09 19:44:40 [14,677,397ms] [Error] [omni.physx.plugin] Cuda context manager error, simulation will be stopped and new cuda context manager will be created.

2024-11-09 19:44:40 [14,677,399ms] [Error] [omni.physx.plugin] PhysX error: Scene state is corrupted. Simulation cannot continue!, FILE /builds/omniverse/physics/physx/source/physx/src/NpScene.cpp, LINE
2876
```

I am curious if there is a max number of contact sensors I can create or how should I resolve this issue? I am using Isaac sim 4.2 and the newest Isaac Lab.

Thanks!

RandomOakForest, 2024-11-12:
This is due to a known issue the team is currently addressing. Thanks for posting this.

supersglzc, 2024-11-13:
Thanks for the reply. I think this is due to the MultiAssetSpawnerCfg. Once you have two rigid objects that are both using this spawner, the error pops.

RandomOakForest, 2024-11-16:
Thanks. We will look into this.

linchangyi1, 2024-11-20:
Hi @supersglzc, I encountered the same issue and would like to kindly ask if you’ve found any alternative solutions to fix it~

linchangyi1, 2024-11-25:
> > Hi @supersglzc, I encountered the same issue and would like to kindly ask if you’ve found any alternative solutions to fix it~
> 
> Unfortunately no. What is your setup?

When a robot has over 256 links, it can't be loaded by the simulation successfully. I think probably the maximal limitation of the link number is 256.

supersglzc, 2025-01-02:
Hi, I would like to follow up on this issue @RandomOakForest . Is the issue being solved in the newest version?

RandomOakForest, 2025-01-14:
@supersglzc, a fix is planned for the 2.0 release. Thanks for following up.

supersglzc, 2025-03-05:
@RandomOakForest Hi I notice that 2.0.1 is already came out. Is the bug fixed in the newest version? Thanks for the update!