isaac-sim/IsaacLab #1403: [Question] How to change the Spot Model to Spot w/ Arm Model?

Jaraxxus-Me, 2024-11-10:
Hi, thanks for the amazing tutorials about training vel control of Spot w/o Arm.
I hope to train a new vel control policy for Spot w/ Arm.
First, I created a custom .usd file for it, adding the arm joints.
Then I tried to modify the SPOT_CFG (extensions/omni.isaac.lab_assets/omni/isaac/lab_assets/spot.py) as follows:
```
# Copyright (c) 2022-2024, The Isaac Lab Project Developers.
# All rights reserved.
#
# SPDX-License-Identifier: BSD-3-Clause

"""Configuration for the Boston Dynamics robot.

The following configuration parameters are available:

* :obj:`SPOT_CFG`: The Spot robot with delay PD and remote PD actuators.
"""

import torch

import omni.isaac.lab.sim as sim_utils
from omni.isaac.lab.actuators import DelayedPDActuatorCfg, RemotizedPDActuatorCfg
from omni.isaac.lab.assets.articulation import ArticulationCfg
from omni.isaac.lab.utils.assets import ISAAC_NUCLEUS_DIR

# Note: This data was collected by the Boston Dynamics AI Institute.
joint_parameter_lookup = torch.tensor([
    [-2.792900, -24.776718, 37.165077],
    [-2.767442, -26.290108, 39.435162],
    [-2.741984, -27.793369, 41.690054],
    [-2.716526, -29.285997, 43.928996],
    [-2.691068, -30.767536, 46.151304],
    [-2.665610, -32.237423, 48.356134],
    [-2.640152, -33.695168, 50.542751],
    [-2.614694, -35.140221, 52.710331],
    [-2.589236, -36.572052, 54.858078],
    [-2.563778, -37.990086, 56.985128],
    [-2.538320, -39.393730, 59.090595],
    [-2.512862, -40.782406, 61.173609],
    [-2.487404, -42.155487, 63.233231],
    [-2.461946, -43.512371, 65.268557],
    [-2.436488, -44.852371, 67.278557],
    [-2.411030, -46.174873, 69.262310],
    [-2.385572, -47.479156, 71.218735],
    [-2.360114, -48.764549, 73.146824],
    [-2.334656, -50.030334, 75.045502],
    [-2.309198, -51.275761, 76.913641],
    [-2.283740, -52.500103, 78.750154],
    [-2.258282, -53.702587, 80.553881],
    [-2.232824, -54.882442, 82.323664],
    [-2.207366, -56.038860, 84.058290],
    [-2.181908, -57.171028, 85.756542],
    [-2.156450, -58.278133, 87.417200],
    [-2.130992, -59.359314, 89.038971],
    [-2.105534, -60.413738, 90.620607],
    [-2.080076, -61.440529, 92.160793],
    [-2.054618, -62.438812, 93.658218],
    [-2.029160, -63.407692, 95.111538],
    [-2.003702, -64.346268, 96.519402],
    [-1.978244, -65.253670, 97.880505],
    [-1.952786, -66.128944, 99.193417],
    [-1.927328, -66.971176, 100.456764],
    [-1.901870, -67.779457, 101.669186],
    [-1.876412, -68.552864, 102.829296],
    [-1.850954, -69.290451, 103.935677],
    [-1.825496, -69.991325, 104.986988],
    [-1.800038, -70.654541, 105.981812],
    [-1.774580, -71.279190, 106.918785],
    [-1.749122, -71.864319, 107.796478],
    [-1.723664, -72.409088, 108.613632],
    [-1.698206, -72.912567, 109.368851],
[... 181 lines cut ...]
        self.scene.robot = SPOT_ARM_CFG.replace(prim_path="{ENV_REGEX_NS}/Robot")
```
Then I tried to start training, but got these two errors:
Before Simulator Starts:
```
Traceback (most recent call last):
  File "/IsaacPlan/ext/extensions/omni.isaac.lab/omni/isaac/lab/sensors/sensor_base.py", line 65, in <lambda>
    lambda event, obj=weakref.proxy(self): obj._initialize_callback(event),
  File "/IsaacPlan/ext/extensions/omni.isaac.lab/omni/isaac/lab/sensors/sensor_base.py", line 275, in _initialize_callback
    self._initialize_impl()
  File "/IsaacPlan/ext/extensions/omni.isaac.lab/omni/isaac/lab/sensors/contact_sensor/contact_sensor.py", line 262, in _initialize_impl
    raise RuntimeError(
RuntimeError: Sensor at path '/World/envs/env_.*/Robot/.*' could not find any bodies with contact reporter API.
HINT: Make sure to enable 'activate_contact_sensors' in the corresponding asset spawn configuration.
```
After Simulator starts:
```
  File "/IsaacPlan/ext/extensions/omni.isaac.lab/omni/isaac/lab/managers/manager_base.py", line 246, in _resolve_common_term_cfg
    raise ValueError(f"Error while parsing '{term_name}:{key}'. {e}")
ValueError: Error while parsing 'add_base_mass:asset_cfg'. Not all regular expressions are matched! Please check that the regular expressions are correct: 
        body: []
Available strings: ['arm0_link_sh1', 'arm0_link_sh0', 'arm0_link_el0', 'base', 'arm0_link_el1', 'fl_hip', 'fr_hip', 'hl_hip', 'hr_hip', 'arm0_link_wr0', 'fl_uleg', 'fr_uleg', 'hl_uleg', 'hr_uleg', 'arm0_link_wr1', 'fl_lleg', 'fr_lleg', 'hl_lleg', 'hr_lleg', 'arm0_link_fngr']
```
Any suggestions/pointers for these?
Thanks.

StrainFlow, 2024-11-12:
For the first error, it looks like you may have changed prim paths of some of the sensors and joints when you changed assets. 

For the second error it looks like the regular expression in the environment setup doesn't match your gripper assets and doesn't know what to do with them.

I'm working on something similar at the moment and I'll keep responding as I learn more.

Jaraxxus-Me, 2024-11-12:
Thanks a lot for pointing out the problem, we are trying to fix it!

StrainFlow, 2024-11-13:
Was that the issue?

meghbhalerao, 2024-11-13:
Hi @Jaraxxus-Me , my issue is not realated to this, but I noticed that you are working with USD files, and I was wondering how you went about modifying and viewing the USD files? I am facing lot of trouble installing Usd by following the instructions in the repo here - https://github.com/PixarAnimationStudios/OpenUSD - if you can point me to any place where you downloaded openusd from that would be super helpful, thanks!

kovlo, 2025-01-10:
Dear @Jaraxxus-Me and @StrainFlow  Did you manage to use the Spot w/ Arm in the simulations?
I added the SPOT_ARM_CFG as shown above, but with using the built in spot_with_arm.usd file. I modified flat env cfg as shown above, but I got the following errors:

```
[INFO]: Starting the simulation. This may take a few seconds. Please wait...
2025-01-10 09:12:05 [14,540ms] [Warning] [omni.physx.plugin] Joint friction attribute is only applied for joints in articulations. (/World/envs/env_0/Robot/base/arm0_sh0)
2025-01-10 09:12:05 [14,540ms] [Warning] [omni.physx.plugin] Joint friction attribute is only applied for joints in articulations. (/World/envs/env_0/Robot/arm0_link_sh0/arm0_sh1)
2025-01-10 09:12:05 [14,541ms] [Warning] [omni.physx.plugin] Joint friction attribute is only applied for joints in articulations. (/World/envs/env_0/Robot/arm0_link_sh1/arm0_el0)
2025-01-10 09:12:05 [14,541ms] [Warning] [omni.physx.plugin] Joint friction attribute is only applied for joints in articulations. (/World/envs/env_0/Robot/arm0_link_el0/arm0_el1)
2025-01-10 09:12:05 [14,541ms] [Warning] [omni.physx.plugin] Joint friction attribute is only applied for joints in articulations. (/World/envs/env_0/Robot/arm0_link_el1/arm0_wr0)
2025-01-10 09:12:05 [14,541ms] [Warning] [omni.physx.plugin] Joint friction attribute is only applied for joints in articulations. (/World/envs/env_0/Robot/arm0_link_wr0/arm0_wr1)
2025-01-10 09:12:05 [14,541ms] [Warning] [omni.physx.plugin] Joint friction attribute is only applied for joints in articulations. (/World/envs/env_0/Robot/arm0_link_wr1/arm0_f1x)
2025-01-10 09:12:06 [15,425ms] [Error] [omni.physx.plugin] PhysX error: Fetching GPU Narrowphase failed! 700
, FILE /builds/omniverse/physics/physx/source/gpunarrowphase/src/PxgNarrowphaseCore.cpp, LINE 1367
2025-01-10 09:12:06 [15,425ms] [Error] [omni.physx.plugin] Cuda context manager error, simulation will be stopped and new cuda context manager will be created.
2025-01-10 09:12:06 [15,426ms] [Error] [omni.physx.plugin] PhysX error: SynchronizeStreams cuEventRecord failed with error 700
, FILE /builds/omniverse/physics/physx/source/gpucommon/include/PxgCudaUtils.h, LINE 57
2025-01-10 09:12:06 [15,426ms] [Error] [omni.physx.plugin] Cuda context manager error, simulation will be stopped and new cuda context manager will be created.
2025-01-10 09:12:06 [15,426ms] [Error] [omni.physx.plugin] PhysX error: SynchronizeStreams cuStreamWaitEvent failed with error 700
, FILE /builds/omniverse/physics/physx/source/gpucommon/include/PxgCudaUtils.h, LINE 63
2025-01-10 09:12:06 [15,426ms] [Error] [omni.physx.plugin] Cuda context manager error, simulation will be stopped and new cuda context manager will be created.
2025-01-10 09:12:06 [15,440ms] [Error] [omni.physx.plugin] PhysX error: PxgCudaDeviceMemoryAllocator failed to allocate memory 67108864 bytes! Result = 700, FILE /builds/omniverse/physics/physx/source/gpucommon/src/PxgCudaMemoryAllocator.cpp, LINE 140
2025-01-10 09:12:06 [15,440ms] [Error] [omni.physx.plugin] Cuda context manager error, simulation will be stopped and new cuda context manager will be created.
2025-01-10 09:12:06 [15,461ms] [Error] [omni.physx.plugin] PhysX error: SynchronizeStreams cuEventRecord failed with error 700
, FILE /builds/omniverse/physics/physx/source/gpucommon/include/PxgCudaUtils.h, LINE 57
2025-01-10 09:12:06 [15,461ms] [Error] [omni.physx.plugin] Cuda context manager error, simulation will be stopped and new cuda context manager will be created.
2025-01-10 09:12:06 [15,461ms] [Error] [omni.physx.plugin] PhysX error: SynchronizeStreams cuStreamWaitEvent failed with error 700
, FILE /builds/omniverse/physics/physx/source/gpucommon/include/PxgCudaUtils.h, LINE 63
2025-01-10 09:12:06 [15,461ms] [Error] [omni.physx.plugin] Cuda context manager error, simulation will be stopped and new cuda context manager will be created.
2025-01-10 09:12:06 [15,462ms] [Error] [omni.physx.plugin] PhysX error: SynchronizeStreams cuEventRecord failed with error 700
, FILE /builds/omniverse/physics/physx/source/gpucommon/include/PxgCudaUtils.h, LINE 57
2025-01-10 09:12:06 [15,462ms] [Error] [omni.physx.plugin] Cuda context manager error, simulation will be stopped and new cuda context manager will be created.
2025-01-10 09:12:06 [15,462ms] [Error] [omni.physx.plugin] PhysX error: SynchronizeStreams cuStreamWaitEvent failed with error 700
, FILE /builds/omniverse/physics/physx/source/gpucommon/include/PxgCudaUtils.h, LINE 63
2025-01-10 09:12:06 [15,462ms] [Error] [omni.physx.plugin] Cuda context manager error, simulation will be stopped and new cuda context manager will be created.
2025-01-10 09:12:06 [15,474ms] [Warning] [omni.physx.plugin] PhysX warning: /builds/omniverse/physics/physx/source/gpucommon/src/PxgCudaMemoryAllocator.cpp, FILE /builds/omniverse/physics/physx/source/gpucommon/src/PxgCudaMemoryAllocator.cpp, LINE 68
```

Hasipasi, 2025-01-13:
@kovlo Same here... I'll let you know if I find something.

closed 2025-01-14 by RandomOakForest

mschweig, 2025-02-01:
Hi, anyone already integrated Spot with Arm? I assume the Joint Lookup Table needs to be extended with BD AI Institute?

kovlo, 2025-02-05:
> [@kovlo](https://github.com/kovlo), [@Hasipasi](https://github.com/Hasipasi), those errors may be related to a PhysX extension bug. Could you please open a new _bug report_ so we can investigate? Please use Isaac Sim 4.2 and Isaac Lab 1.4.0. Thanks.

FYI: I was running Isaac Sim 4.2 and Isaac Lab 1.4.0.
I check the new fresh versions (4.5 and 2.0) and I'll come back with my findings either here or to a new issue.

mschweig, 2025-02-05:
> FYI: I was running Isaac Sim 4.2 and Isaac Lab 1.4.0. I check the new fresh versions (4.5 and 2.0) and I'll come back with my findings either here or to a new issue.

I already tested it with 4.5 and 2.0 and still getting the PhysX errors.

StrainFlow, 2025-02-12:
Can you get either a spot or a standalone arm working? In other words is this problem specific to a spot with an arm?

eygu, 2025-04-07:
We had the same issue as you with the regular expressions list being empty, we made a workaround by copying the arm from the spot_arm.usd model to the spot.usd that was working and got it working that way. The file is in our repo: [spot_fixed.usda](https://github.com/quacklemtduck/spot-stuff/blob/working/source/spot_stuff/spot_stuff/tasks/msc/spot/assets/spot_fixed.usda) 
We also had to add each joint actuators manually and set the parameters individually (seen in spot_arm.py)

YuHoChau, 2025-04-18:
> We had the same issue as you with the regular expressions list being empty, we made a workaround by copying the arm from the spot_arm.usd model to the spot.usd that was working and got it working [...]

Hi, thank you for sharing your work! I came across your codebase [here](https://github.com/quacklemtduck/spot-stuff/blob/171c4b9f267d2b368539cdc71fbda829f6e87762/source/spot_stuff/spot_stuff/tasks/msc/spot/flat_env_cfg.py), and I was wondering are you also working on loco-manipulation tasks? I’m currently working on a similar project using a different quadruped robot with an arm, but I’ve run into issues computing the arm Jacobian from the whole body using `robot.root_physx_view.get_jacobians()`. Specifically, if I fix the base of the dog the IK of the arm can be solved with correct Jacobian but if make it as a float base, even the Jacobian matrix for the whole-body is incorrect with many zeros. Have you encountered a similar issue in your work or did you use Jacobian for IK? If so, I’d really appreciate any insight on how you approached or resolved it. Thank you very much.

eygu, 2025-04-23:
> Hi, thank you for sharing your work! I came across your codebase [...]

Hi, we are not using jacobians or IK. We are on purpose only using reinforcement learning for everything, so I'm not sure that we can help. We have encountered weird values at times though, and maybe our fix can help you. There seems to be some form of issue when reading parameters for the actuators in the USD model, and we would get weird movements and results when we relied on that. Our fix was to just manually define the actuator parameters for each joint individually. We used the OpenUSD tools to inspect the model and get these values.

YuHoChau, 2025-04-24:
> Hi, we are not using jacobians or IK. We are on purpose only using reinforcement learning for everything, so I'm not sure that we can help. We have encountered weird values at times though, and [...]

Thank you for your help!