isaac-sim/IsaacLab #1404: [Question] Failed to Run IsaacLab with 4090 on Ubuntu 20.04

YuHoChau, 2024-11-11:
### Question

Hi, I've been trying to install IsaacLab on my workstation running Ubuntu 20.04. I followed all the steps using the [binaries installation method](https://isaac-sim.github.io/IsaacLab/main/source/setup/installation/binaries_installation.html). I successfully verified the installation on another PC with an RTX 3080 GPU, so I believe the method should be correct.

However, while some demos (such as the showroom demos featuring quadrupeds) run successfully, others, including demos with RL wrappers (like training with rsl_rl on the Isaac-Reach-Franka-v0 task), fail. Specifically, for some showroom demos, the Isaac Sim window crashes. For rsl_rl demos, it reports errors:

```
failed call to posix_fadvise(-9223232211531366400, -1389729927752357637)ret=22, errno=0 'Success'
failed call to posix_madvise(-9223232211531366400, -1389729927752357637)ret=22, errno=0 'Success'
LLVM ERROR: out of memory
```

For reference, I am using the recommended driver vision 535.192.03 with CUDA 11.8, on a 4090 + 13900k workstation with 32G RAM. I have tested Isaac Sim and be sure it works well so I assume the problem is from IsaacLab. If someone had the similar questions and fixed it, please let me know. Thank you for your help.

RandomOakForest, 2024-11-12:
This is due to a known issue the team is addressing. Will tag for fixing. Thanks for posting this.

RandomOakForest, 2024-11-16:
It may be related to a driver issue. The team is investigating this. Thanks.

yinzikang, 2024-11-21:
For scripts under standalone/demos, only the `quadcopter` and the `deformables` can work. These failed ones will pop up the Isaac Sim window, but nothing is displayed.

RandomOakForest, 2025-02-11:
This could not be reproduced. The next step would be to try this in the latest Isaac Sim and Isaac Lab releases. If you still face this issue please post a new _Bug Report_. Thanks.

closed 2025-02-11 by RandomOakForest