isaac-sim/IsaacLab #1405: Issues with Actuator Configuration and Commanding Robot Movements in Simulation

sinaqahremani, 2024-11-11:
Hi,

I am working on developing a reinforcement learning (RL) policy for a mobile manipulator setup that includes a Ridgeback base and a Kuka arm. I have provided a `usda` file that describes the arm and have defined a configuration (`CFG`) for the robot, as shown below:

```python

RIDGEBACK_KUKA_CFG = ArticulationCfg(
    spawn = sim_utils.UsdFileCfg(
        usd_path=str(usd_path),
        articulation_props=sim_utils.ArticulationRootPropertiesCfg(enabled_self_collisions=False),
        activate_contact_sensors=False,
        rigid_props=sim_utils.RigidBodyPropertiesCfg(
            disable_gravity=False,
            max_depenetration_velocity=5.0,
        ),
    ),
    init_state=ArticulationCfg.InitialStateCfg(
        pos=(0.0, 0.0, 0.0),
        joint_pos={
            "front_rocker": 0.0,
            # base
            "front_left_wheel": 0.0,
            "front_right_wheel": 0.0,

            "rear_left_wheel": 0.0,
            "rear_right_wheel": 0.0,
  

            # ewellix
            "ewellix_lift_top_joint": 0.0,

            # Kuka arm
            "iiwa_joint_1": 0.0,
            "iiwa_joint_2": -1.7,
            "iiwa_joint_3": 0.0,
            "iiwa_joint_4": -0.45,
            "iiwa_joint_5": 0.0,
            "iiwa_joint_6": .20,
            "iiwa_joint_7": 0.0,
        },
        joint_vel={".*": 0.0}
    ),
    actuators={
        "mobile_base_front": ImplicitActuatorCfg(
            joint_names_expr=["front_.*"],
            # velocity_limit=500.0,
            # effort_limit=1000.0,
            # stiffness=0.0,
            # damping=10000000.0,
        ),
        "mobile_base_rear": ImplicitActuatorCfg(
            joint_names_expr=["rear_.*"],
            # velocity_limit=500.0,
            # effort_limit=1000.0,
            # stiffness=0.0,
            # damping=10000000.0,
        ),
        "arm_actuator": ImplicitActuatorCfg(
            joint_names_expr=["iiwa_joint_.*"], 
            effort_limit=300.0, 
            velocity_limit=100.0, 
            stiffness=10000000.0, 
            damping=10000000.0
        ),
        "lifting": ImplicitActuatorCfg(
            joint_names_expr=["ewellix_lift_top_joint"], 
            effort_limit=100000.0, 
            velocity_limit=100.0, 
            stiffness=10000000.0, 
            damping=0.0
        ),
    },
)

```

Additionally, I wrote a simple script to test the setup by moving the arm and the base. Here’s the relevant part of the script:

```python

@configclass
class TableTopSceneCfg(InteractiveSceneCfg):
    """Configuration for a cart-pole scene."""

    # ground plane
    ground = AssetBaseCfg(
        prim_path="/World/defaultGroundPlane",
        spawn=sim_utils.GroundPlaneCfg(),
        init_state=AssetBaseCfg.InitialStateCfg(pos=(0.0, 0.0, 0.0)),
    )

    # lights
    dome_light = AssetBaseCfg(
        prim_path="/World/Light", spawn=sim_utils.DomeLightCfg(intensity=3000.0, color=(0.75, 0.75, 0.75))
    )

    # articulation
    robot = RIDGEBACK_KUKA_CFG.replace(prim_path="{ENV_REGEX_NS}/Robot")

def run_simulator(sim: sim_utils.SimulationContext, scene: InteractiveScene):
    """Runs the simulation loop."""

    robot = scene["robot"]
    # Define simulation stepping
    sim_dt = sim.get_physics_dt()
    print(sim_dt)
    count = 0
    # Simulation loop

    while simulation_app.is_running():
            
        joint_pos, joint_vel = robot.data.joint_pos.clone(), robot.data.joint_vel.clone()

        if count % 301 == 0:
            print("joint_pose")
            print(joint_pos)

            print("Joint vel")
            print(joint_vel)

        if count > 200:
            joint_vel = torch.zeros_like(joint_vel)

            joint_vel[0, 1] = 1000.0
            joint_vel[0, 2] = 1000.0
            joint_vel[0, 3] = 1000.0
            joint_vel[0, 4] = 1000.0

            joint_pos = torch.zeros_like(joint_pos)
            joint_pos[0:, 7] = -1.2
            joint_pos[0:, 9] =  1.0
            robot.write_joint_state_to_sim(joint_pos, joint_vel)
            # print("joint_pose")
            # print(joint_pos)
            # print("Joint vel")
            # print(joint_vel)
            # robot.set_joint_position_target(joint_pos)
            # robot.set_joint_velocity_target(joint_vel)

        # clear internal buffers
        robot.reset()

        scene.write_data_to_sim()
        # Perform step
        sim.step()
        # Increment counter
        count += 1
        # Update buffers
        robot.update(sim_dt)

def main():
    """Main function."""
    # Load kit helper
    sim_cfg = sim_utils.SimulationCfg(dt=0.01)
    sim = sim_utils.SimulationContext(sim_cfg)
    # Set main camera
    sim.set_camera_view([2.5, 2.5, 2.5], [0.0, 0.0, 0.0])
    # Design scene
    scene_cfg = TableTopSceneCfg(num_envs=1, env_spacing=5.0)
    scene = InteractiveScene(scene_cfg)
    # Play the simulator
    sim.reset()
    # Now we are ready!
    print("[INFO]: Setup complete...")
    # Run the simulator
    run_simulator(sim, scene)

```

I am encountering several issues with my simulation setup:

1. Even without explicitly setting joint velocities or positions, the robotic arm does not initialize in the defined configuration specified in `RIDGEBACK_KUKA_CFG`.
2. When calling the following methods:
   ```python
   robot.set_joint_position_target(joint_pos)
   robot.set_joint_velocity_target(joint_vel)
   ```
   neither the velocity nor the position of the joints changes.
3. When executing `robot.write_joint_state_to_sim(joint_pos, joint_vel)`, only the arm moves, while the base remains stationary.

I would greatly appreciate any insights or suggestions to resolve these issues. Thank you for your help!

RandomOakForest, 2024-11-16:
Thanks for posting this. There are a couple of items in your questions we are still reviewing with our engineering team. Will aim to post a suggested direction next week.

Afshin1997, 2024-12-01:
Hi Sina,
The problem that you are facing now seems to be due to the high stiffness, damping and joint_velocities you have chosen for your manipulator. I faced this issue on another robot. Start with lower values like 200 for stiffness and 30 for damping if you are working with implicit actuators.

RandomOakForest, 2024-12-16:
Our apologies following up has taking more than we anticipated. Thanks for following up, @Afshin1997. As Afshin points out, the stiffness and damping values are very large. Is the current version of Isaac Lab you are using 1.3? Thanks

RandomOakForest, 2025-01-02:
Following up on this, let us know if you still have this issue despite the suggestions above, and if you have been able to use release 1.4.0. Thanks.

closed 2025-01-15 by RandomOakForest