isaac-sim/IsaacLab #1406: [Question] Question title

GordonLuQwQ, 2024-11-11:
### Question
2024-11-11 05:03:17 [11,730ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-11-11 05:03:17 [11,730ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-11-11 05:03:17 [11,730ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-11-11 05:03:17 [11,730ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-11-11 05:03:17 [11,730ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-11-11 05:03:17 [11,730ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-11-11 05:03:17 [11,730ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-11-11 05:03:17 [11,730ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-11-11 05:03:17 [11,730ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-11-11 05:03:17 [11,730ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-11-11 05:03:17 [11,730ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-11-11 05:03:17 [11,730ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-11-11 05:03:17 [11,730ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Core' for removal
2024-11-11 05:03:17 [11,879ms] [Warning] [carb] Recursive unloadAllPlugins() detected!
I have this error when i try to run a trained result, i saw on the website someone has solved it by downloading the assets pack, but i don't think they are available on the launcher anymore, please help

RandomOakForest, 2024-11-12:
This may be due to an issue currently being addressed by the team. Thanks for posting this.

RandomOakForest, 2024-11-12:
Could you please confirm if this is producing a crash? Are there any other errors before these warnings? Thanks.

DSIsaac, 2024-11-14:
> Could you please confirm if this is producing a crash? Are there any other errors before these warnings? Thanks.

I have met the same question. The whole message is as follows
2024-11-14 05:18:40 [3,415ms] [Warning] [omni.replicator.core.scripts.annotators] Annotator PostProcessDispatch is already registered, overwriting annotator template
2024-11-14 05:23:51 [314,430ms] [Warning] [omni.usd] Warning: in _ReportErrors at line 2890 of C:\g\97199581\USD\pxr\usd\usd\stage.cpp -- In </World/defaultGroundPlane>: Could not open asset @http://omniverse-content-production.s3-us-west-2.amazonaws.com/Assets/Isaac/4.2/Isaac/Environments/Grid/default_environment.usd@ for reference introduced by @anon:0000021AC2954220:World0.usd@</World/defaultGroundPlane>. (recomposing stage on stage @anon:0000021AC2954220:World0.usd@ <0000021AA6E6DE70>)

Traceback (most recent call last):
  File "C:\Software\IsaacLab\source\standalone\demos\deformables.py", line 199, in <module>
    main()
  File "C:\Software\IsaacLab\source\standalone\demos\deformables.py", line 186, in main
    scene_entities, scene_origins = design_scene()
  File "C:\Software\IsaacLab\source\standalone\demos\deformables.py", line 62, in design_scene
    cfg_ground.func("/World/defaultGroundPlane", cfg_ground)
  File "c:\software\isaaclab\source\extensions\omni.isaac.lab\omni\isaac\lab\sim\spawners\from_files\from_files.py", line 145, in spawn_ground_plane
    collision_prim_path = prim_utils.get_prim_path(
  File "c:/software/isaaclab/_isaac_sim/exts/omni.isaac.core/omni/isaac/core/utils/prims.py", line 612, in get_prim_path    return prim.GetPath()
AttributeError: 'NoneType' object has no attribute 'GetPath'
2024-11-14 05:23:51 [314,546ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-11-14 05:23:51 [314,546ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-11-14 05:23:51 [314,547ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-11-14 05:23:51 [314,548ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-11-14 05:23:51 [314,548ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-11-14 05:23:51 [314,548ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-11-14 05:23:51 [314,549ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-11-14 05:23:51 [314,549ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-11-14 05:23:51 [314,549ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-11-14 05:23:51 [314,550ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-11-14 05:23:51 [314,550ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-11-14 05:23:51 [314,550ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-11-14 05:23:51 [314,551ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Core' for removal2024-11-14 05:23:51 [314,674ms] [Warning] [carb] Recursive unloadAllPlugins() detected!

RandomOakForest, 2024-11-16:
This points to an initial setup and install issue. Are you using the latest Isaac Sim/Lab? Do you have previous Omniverse installations? In some cases doing a [launcher clean up](https://docs.omniverse.nvidia.com/utilities/latest/cleanup-tool.html) has helped.

RandomOakForest, 2024-12-10:
We hope the previous suggestion helped, I will close this for now. You may reopen this topic, though we will appreciate further information for us to investigate if the issue has persisted. Thank you.

closed 2024-12-10 by RandomOakForest

LittlePotatoChip, 2024-12-23:
> 'Replicator:Annotators' for removal
> 2024-11-11 05:03:17 [11,730ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
> 2024-11-11 05:03:17 [11,730ms] [...]

same questions，I tried to  run a demo with this:
`./isaaclab.sh -p source/standalone/tutorials/00_sim/spawn_prims.py`

`(isaaclab) potato@potato-MS-7E13:~/workplace/IsaacLab$ ./isaaclab.sh -p source/standalone/tutorials/00_sim/spawn_prims.py
[INFO] Using python from: /home/potato/miniforge3/envs/isaaclab/bin/python     
[WARN][AppLauncher]: There are no arguments attached to the ArgumentParser object. If you have your own arguments, please load your own arguments before calling the `AppLauncher.add_app_launcher_args` method. This allows the method to check the validity of the arguments and perform checks for argument names.
[INFO][AppLauncher]: Loading experience file: /home/potato/workplace/IsaacLab/source/apps/isaaclab.python.kit
[Warning] [omni.isaac.kit.simulation_app] Modules: ['omni.kit_app'] were loaded before SimulationApp was started and might not be loaded correctly.
[Warning] [omni.isaac.kit.simulation_app] Please check to make sure no extra omniverse or pxr modules are imported before the call to SimulationApp(...)
Loading user config located at: '/home/potato/miniforge3/envs/isaaclab/lib/python3.10/site-packages/omni/data/Kit/Isaac-Sim/4.2/user.config.json'
[Info] [carb] Logging to file: /home/potato/miniforge3/envs/isaaclab/lib/python3.10/site-packages/omni/logs/Kit/Isaac-Sim/4.2/kit_20241223_211801.log
2024-12-23 13:18:01 [0ms] [Warning] [omni.kit.app.plugin] No crash reporter present, dumps uploading isn't available.

|---------------------------------------------------------------------------------------------|
| Driver Version: 550.120       | Graphics API: Vulkan
|=============================================================================================|
| GPU | Name                             | Active | LDA | GPU Memory | Vendor-ID | LUID       |
|     |                                  |        |     |            | Device-ID | UUID       |
|     |                                  |        |     |            | Bus-ID    |            |
|---------------------------------------------------------------------------------------------|
| 0   | NVIDIA GeForce RTX 4060 Ti       | Yes: 0 |     | 16380   MB | 10de      | 0          |
|     |                                  |        |     |            | 2805      | 52e3204a.. |
|     |                                  |        |     |            | 1         |            |
|=============================================================================================|
| OS: 24.04.1 LTS (Noble Numbat) ubuntu, Version: 24.04.1, Kernel: 6.8.0-49-generic
| XServer Vendor: The X.Org Foundation, XServer Version: 12101011 (1.21.1.11)
| Processor: 13th Gen Intel(R) Core(TM) i5-13600KF | Cores: 14 | Logical: 20
|---------------------------------------------------------------------------------------------|
| Total Memory (MB): 31930 | Free Memory: 26735
| Total Page/Swap (MB): 3814 | Free Page/Swap: 3813
|---------------------------------------------------------------------------------------------|
2024-12-23 13:18:02 [423ms] [Warning] [gpu.foundation.plugin] IOMMU is enabled.
2024-12-23 13:18:06 [4,556ms] [Warning] [omni.replicator.core.scripts.annotators] Annotator PostProcessDispatch is already registered, overwriting annotator template
2024-12-23 13:18:06 [4,711ms] [Warning] [omni.kit.widget.cache_indicator.cache_state_menu] Unable to detect Omniverse Cache Server. File /home/potato/.nvidia-omniverse/config/omniverse.toml is not found. Consider installing it for better IO performance.
2024-12-23 13:20:28 [146,595ms] [Warning] [omni.usd] Warning: in _ReportErrors at line 2890 of /builds/omniverse/usd-ci/USD/pxr/usd/usd/stage.cpp -- In </World/defaultGroundPlane>: Could not open asset @http://omniverse-content-production.s3-us-west-2.amazonaws.com/Assets/Isaac/4.2/Isaac/Environments/Grid/default_environment.usd@ for reference introduced by @anon:0x5c0d2a00d3c0:World0.usd@</World/defaultGroundPlane>. (recomposing stage on stage @anon:0x5c0d2a00d3c0:World0.usd@ <0x5c0d282fd970>)

Traceback (most recent call last):
  File "/home/potato/workplace/IsaacLab/source/standalone/tutorials/00_sim/spawn_prims.py", line 116, in <module>
    main()
  File "/home/potato/workplace/IsaacLab/source/standalone/tutorials/00_sim/spawn_prims.py", line 101, in main
    design_scene()
  File "/home/potato/workplace/IsaacLab/source/standalone/tutorials/00_sim/spawn_prims.py", line 44, in design_scene
    cfg_ground.func("/World/defaultGroundPlane", cfg_ground)
  File "/home/potato/workplace/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/sim/spawners/from_files/from_files.py", line 145, in spawn_ground_plane
    collision_prim_path = prim_utils.get_prim_path(
  File "/home/potato/miniforge3/envs/isaaclab/lib/python3.10/site-packages/isaacsim/exts/omni.isaac.core/omni/isaac/core/utils/prims.py", line 612, in get_prim_path
    return prim.GetPath()
AttributeError: 'NoneType' object has no attribute 'GetPath'
2024-12-23 13:20:28 [146,661ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-12-23 13:20:28 [146,661ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
[... cut ...]
2024-12-23 13:20:28 [146,661ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-12-23 13:20:28 [146,661ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-12-23 13:20:28 [146,661ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-12-23 13:20:28 [146,661ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Core' for removal
2024-12-23 13:20:28 [146,702ms] [Warning] [carb] Recursive unloadAllPlugins() detected!`

Have you sovle this?