isaac-sim/IsaacLab #1409: [Question] Encountering Unknown Error: [Error] [omni.physx.cooking.plugin] opRes == OperationResult::SUCCESS is false.

H-Hisamichi, 2024-11-11:
Hello everyone,

I encountered an unknown error while training my original Hexapod using reinforcement learning. 

First, I get this warning: `[Warning] [omni.ujitso] UJITSO: Build storage validation failed for Storage status: 1`, and then the error `[Error] [omni.physx.cooking.plugin] opRes == OperationResult::SUCCESS is false.` appears.

Sometimes this error causes a buffer error, but other times the training continues without any issues. I couldn't find any information about this error.
Does anyone know what this means?

Thank you!

RandomOakForest, 2024-11-12:
Could you please confirm the version of Isaac Lab you are using? Have you tried this in the 1.2 release? Thanks!

H-Hisamichi, 2024-11-13:
Hello @RandomOakForest ,

I'm using Isaac Lab v1.2.0 and Isaac Sim v4.2.0.
Thanks again!

RandomOakForest, 2024-11-16:
Thank you for submitting this. We have take note of this error. If it is causing a crash on your side, please submit a bug report. Thanks.

H-Hisamichi, 2024-11-17:
Hello @RandomOakForest,

Thank you for your response.
The current error code does not cause a crash. However, I am not sure if this error affects policy.
I will provide more information about my environment.
Isaac Sim: 4.2.0
Isaac Lab: 1.2.0
RL code: based on anymal_c(direct), modified for original hexapod.

The display of error codes has also been changed. The cause is unknown.
Example of error code pattern 1:
```
[INFO] Using python from: /home/robot/anaconda3/envs/iccar2025/bin/python
[INFO][AppLauncher]: Loading experience file: /home/robot/IsaacLab_v1.2.0/IsaacLab/source/apps/isaaclab.python.headless.rendering.kit
Loading user config located at: '/home/robot/.local/share/ov/pkg/isaac-sim-4.2.0/kit/data/Kit/Isaac-Sim/4.2/user.config.json'
[Info] [carb] Logging to file: /home/robot/.local/share/ov/pkg/isaac-sim-4.2.0/kit/logs/Kit/Isaac-Sim/4.2/kit_20241117_101409.log
2024-11-17 01:14:10 [0ms] [Warning] [omni.kit.app.plugin] No crash reporter present, dumps uploading isn't available.
2024-11-17 01:14:10 [673ms] [Warning] [carb.windowing-glfw.plugin] GLFW initialization failed.
2024-11-17 01:14:10 [673ms] [Warning] [carb] Failed to startup plugin carb.windowing-glfw.plugin (interfaces: [carb::windowing::IGLContext v1.0],[carb::windowing::IWindowing v1.4]) (impl: carb.windowing-glfw.plugin)
2024-11-17 01:14:10 [701ms] [Warning] [omni.platforminfo.plugin] failed to open the default display.  Can't verify X Server version.

|---------------------------------------------------------------------------------------------|
| Driver Version: 535.183.01    | Graphics API: Vulkan
|=============================================================================================|
| GPU | Name                             | Active | LDA | GPU Memory | Vendor-ID | LUID       |
|     |                                  |        |     |            | Device-ID | UUID       |
|     |                                  |        |     |            | Bus-ID    |            |
|---------------------------------------------------------------------------------------------|
| 0   | NVIDIA RTX A4000                 | Yes: 0 |     | 16622   MB | 10de      | 0          |
|     |                                  |        |     |            | 24b0      | 44ccd3c0.. |
|     |                                  |        |     |            | 6         |            |
|=============================================================================================|
| OS: 22.04.5 LTS (Jammy Jellyfish) ubuntu, Version: 22.04.5, Kernel: 6.8.0-48-generic
| Processor: AMD Ryzen 7 2700 Eight-Core Processor | Cores: 8 | Logical: 16
|---------------------------------------------------------------------------------------------|
| Total Memory (MB): 48078 | Free Memory: 41052
| Total Page/Swap (MB): 2047 | Free Page/Swap: 2047
|---------------------------------------------------------------------------------------------|
2024-11-17 01:14:11 [1,319ms] [Warning] [gpu.foundation.plugin] IOMMU is enabled.
2024-11-17 01:14:12 [2,183ms] [Warning] [carb.windowing-glfw.plugin] GLFW initialization failed.
2024-11-17 01:14:12 [2,183ms] [Warning] [carb] Failed to startup plugin carb.windowing-glfw.plugin (interfaces: [carb::windowing::IGLContext v1.0],[carb::windowing::IWindowing v1.4]) (impl: carb.windowing-glfw.plugin)
2024-11-17 01:14:12 [2,193ms] [Warning] [carb.windowing-glfw.plugin] GLFW initialization failed.
2024-11-17 01:14:12 [2,193ms] [Warning] [carb] Failed to startup plugin carb.windowing-glfw.plugin (interfaces: [carb::windowing::IGLContext v1.0],[carb::windowing::IWindowing v1.4]) (impl: carb.windowing-glfw.plugin)
2024-11-17 01:14:12 [2,201ms] [Warning] [carb.windowing-glfw.plugin] GLFW initialization failed.
2024-11-17 01:14:12 [2,201ms] [Warning] [carb] Failed to startup plugin carb.windowing-glfw.plugin (interfaces: [carb::windowing::IGLContext v1.0],[carb::windowing::IWindowing v1.4]) (impl: carb.windowing-glfw.plugin)
2024-11-17 01:14:12 [2,210ms] [Warning] [carb.windowing-glfw.plugin] GLFW initialization failed.
2024-11-17 01:14:12 [2,210ms] [Warning] [carb] Failed to startup plugin carb.windowing-glfw.plugin (interfaces: [carb::windowing::IGLContext v1.0],[carb::windowing::IWindowing v1.4]) (impl: carb.windowing-glfw.plugin)
2024-11-17 01:14:12 [2,212ms] [Warning] [carb.windowing-glfw.plugin] GLFW initialization failed.
2024-11-17 01:14:12 [2,212ms] [Warning] [carb] Failed to startup plugin carb.windowing-glfw.plugin (interfaces: [carb::windowing::IGLContext v1.0],[carb::windowing::IWindowing v1.4]) (impl: carb.windowing-glfw.plugin)
2024-11-17 01:14:12 [2,213ms] [Warning] [carb.windowing-glfw.plugin] GLFW initialization failed.
2024-11-17 01:14:12 [2,213ms] [Warning] [carb] Failed to startup plugin carb.windowing-glfw.plugin (interfaces: [carb::windowing::IGLContext v1.0],[carb::windowing::IWindowing v1.4]) (impl: carb.windowing-glfw.plugin)
2024-11-17 01:14:15 [5,363ms] [Warning] [omni.replicator.core.scripts.annotators] Annotator PostProcessDispatch is already registered, overwriting annotator template
formatted_tuple: (0.5, 0.5, 0.5, 0.5)
[INFO]: Parsing configuration from: omni.isaac.lab_tasks.direct.AT3R.AT3R_rolling_env:AT3RRoughEnvCfg
[INFO]: Parsing configuration from: /home/robot/IsaacLab_v1.2.0/IsaacLab/source/extensions/omni.isaac.lab_tasks/omni/isaac/lab_tasks/direct/AT3R/agents/rl_games_rough_ppo_cfg.yaml
[INFO] Logging experiment in directory: /home/robot/IsaacLab_v1.2.0/IsaacLab/logs/rl_games/AT3R_rough_direct
Setting seed: 42
[... cut ...] e buffer first.
/home/robot/anaconda3/envs/iccar2025/lib/python3.10/site-packages/rl_games/algos_torch/a2c_continuous.py:106: FutureWarning: `torch.cuda.amp.autocast(args...)` is deprecated. Please use `torch.amp.autocast('cuda', args...)` instead.
  with torch.cuda.amp.autocast(enabled=self.mixed_precision):
fps step: 4462 fps step and policy inference: 4405 fps total: 4137 epoch: 1/5000 frames: 0
fps step: 6106 fps step and policy inference: 6017 fps total: 5972 epoch: 2/5000 frames: 24576
```

RandomOakForest, 2024-11-18:
Thanks for posting this. We will review with the team.

RandomOakForest, 2024-12-06:
Some issue related to the PhysX extension have been resolved with the latest release of Isaac Lab 1.3. Could you give this a try and share if you still have this issue? Thanks

H-Hisamichi, 2024-12-07:
Hello @RandomOakForest,

Thank you for your help!
I will try with the latest Isaac Lab.

closed 2024-12-07 by H-Hisamichi