isaac-sim/IsaacLab #1410: [Bug Report] Unable to run the teleop demo via the WebRTC streaming client

mohanksriram, 2024-11-11:
### Bug description
The default teleop demo runs fine natively but when i try to run the teleop demo via WebRTC streaming client I run into
[image]

### Steps to reproduce
run the following command
```
./isaaclab.sh -p source/standalone/environments/teleoperation/teleop_se3_agent.py --task Isaac-Lift-Cube-Franka-IK-Rel-v0 --num_envs 1 --teleop_device keyboard --livestream 2
```

wzds2015, 2024-11-12:
> ### Bug description
> The default teleop demo runs fine natively but when i try to run the teleop demo via WebRTC streaming client I run into <img alt="image" width="833" src="https://private-user-images.githubusercontent.com/22681121/385066338-5510b4c6-d795-4aaa-a252-6482fb64e392.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzEzODM1NzQsIm5iZiI6MTczMTM4MzI3NCwicGF0aCI6Ii8yMjY4MTEyMS8zODUwNjYzMzgtNTUxMGI0YzYtZDc5NS00YWFhLWEyNTItNjQ4MmZiNjRlMzkyLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDExMTIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMTEyVDAzNDc1NFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTlhYmMxMTZmMjhkN2ExNTdkYjk5YzM3YjNiODM0NzUyZTM5N2U2NWM2ZmM0NDVhYjFjNTdhOWQ3N2NlYzJlMmYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.AdeNCG73nlbFCpzV7AK_G8vref7Y6j3zLgzQ7rALHmQ">
> 
> ### Steps to reproduce
> run the following command
> 
> ```
> ./isaaclab.sh -p source/standalone/environments/teleoperation/teleop_se3_agent.py --task Isaac-Lift-Cube-Franka-IK-Rel-v0 --num_envs 1 --teleop_device keyboard --livestream 2
> ```

I have the same question. Are you able to use Streaming Client: --livestream 1?

RandomOakForest, 2024-11-12:
Are you running this in the latest version of Isaac Lab? Thanks

mohanksriram, 2024-11-12:
I'm running Isaac Sim v4.1, Isaac Lab v1.1

RandomOakForest, 2024-11-15:
This may have been resolved with Isaac Sim 4.2.0 and Isaac Lab 1.2.0. Would it be possible to try these? Thanks

vrodriguezf, 2024-11-20:
I'm using Isaac Sim 4.2.0 and IsaacLab 1.2.0 and the issue is there.

RandomOakForest, 2024-11-22:
Thanks, @vrodriguezf. We will investigate.

RandomOakForest, 2025-02-11:
We haven't been able to reproduce this on our side. The next step would be to try Isaac Lab 2.0. Let us know if you still need further help.

AntoineRichard, 2026-08-27:
Thanks for reporting this, @mohanksriram — WebRTC teleop issues are genuinely awkward to debug remotely.

We weren't able to reproduce it at the time, and a lot has changed since Isaac Sim 4.1/4.2: the livestream stack has been reworked and the script has moved to `scripts/environments/teleoperation/teleop_se3_agent.py` (`source/standalone/...` is gone), though `--livestream 2` still selects private WebRTC. Closing as stale — if you or @vrodriguezf still hit this on a current release, please reopen or file a fresh issue with a full log.

---
*🤖 This issue was reviewed as part of an AI-assisted triage of the Isaac Lab backlog, with automated verification against the current codebase. If we got this wrong, please reopen — we'd rather hear about it!*

closed 2026-08-27 by AntoineRichard