isaac-sim/IsaacLab #1415: [Bug Report] Bug title

johnnylu305, 2024-11-13:
### Describe the bug

We tried running IsaacLab on the L40 and L4 machines using the benchmark script in a container. However, we encountered this error: "LLVM ERROR: out of memory. Fatal Python error: Aborted."

We attempted to downgrade the driver to a recommended version following [this](https://github.com/isaac-sim/IsaacLab/issues/429
), but the error persists.

### Steps to reproduce

./isaaclab.sh -p source/standalone/benchmarks/benchmark_rlgames.py --task=Isaac-Cartpole-RGB-Camera-Direct-v0 --headless --enable_cameras

```
LLVM ERROR: out of memory Fatal Python error: Aborted
```

### System Info

Describe the characteristic of your environment:

- Commit: 84b2d2d
- Isaac Sim Version: 4.2.0-rc.18+release.16044.3b2ed111.gl
- OS: RED HAT
- GPU: L4/L40
- CUDA: 12.2
- GPU Driver: 535.171
- Memery: 188GB

### Checklist

- [x] I have checked that there is no similar issue in the repo (**required**)
- [x] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

Add the criteria for which this task is considered **done**. If not known at issue creation time, you can add this once the issue is assigned.

- [ ] Criteria 1
- [ ] Criteria 2

YuHoChau, 2024-11-13:
Same error here.

RandomOakForest, 2024-11-16:
This seems to be specific to that commit. Could you try the [latest release](https://github.com/isaac-sim/IsaacLab/releases/tag/v1.2.0) instead? Thanks!

johnnylu305, 2024-11-16:
@RandomOakForest Sorry, can I know which commit fix the problem? I have tried to reinstall everything several times these two weeks. But, it still does not work.

RandomOakForest, 2025-01-24:
We ran into a couple of holidays and releases, our apologies for the late reply. As of this writing, the latest [Isaac Lab is version 1.4.0, please use this link to get to this tag of the code](https://github.com/isaac-sim/IsaacLab/tree/v1.4.0). Having said that, it seems your Python is running out of memory in a task that is related to the LLVM compiler infrastructure. This could be related to your container setup. Let us know if we can be of further help. Thanks.

YuHoChau, 2025-02-07:
Hello, have you solved this problem? Thanks

RandomOakForest, 2025-02-11:
Thank you for following up. Are you still facing this issue with Isaac Lab 2.0? Thanks.

RandomOakForest, 2025-02-11:
Thanks for trying all this. It seems you are able to run the tutorials without issues. These errors typically happen when your platform is running out of memory. If you only see this issue with your Franka robot, could you please share a full log showing the errors and warning? also please include the latest steps you have used so we can try to reproduce. Thanks.

YuHoChau, 2025-02-13:
Thank for for the comment. Yes I can run the demos from Isaac Sim well but for demos from Isaac Lab, especially using RL libraries like RSL_RL and SKRL, it would report the "`LLVM ERROR`". I once tried to add the RAM up to 64GB still it has the same problem, which I doubt this could be really "out of memory". I have the detailed log here: https://forums.developer.nvidia.com/t/failed-to-run-isaaclab-with-4090-on-ubuntu-20-04/312865. After update to Lab 2.0 version, on my Ubuntu 20, it reports the error: 
```
(env_isaaclab) yuhao@yuhao:~/IsaacLab$ ./isaaclab.sh -p scripts/reinforcement_learning/rsl_rl/train.py --task Isaac-Reach-Franka-v0 --headless
[INFO] Using python from: /home/yuhao/anaconda3/envs/env_isaaclab/bin/python                                       
[INFO][AppLauncher]: Loading experience file: /home/yuhao/IsaacLab/apps/isaaclab.python.headless.kit
Loading user config located at: '/home/yuhao/Downloads/isaac-sim-standalone@4.5.0/kit/data/Kit/Isaac-Sim/4.5/user.config.json'
[Info] [carb] Logging to file: /home/yuhao/Downloads/isaac-sim-standalone@4.5.0/kit/logs/Kit/Isaac-Sim/4.5/kit_20250213_151954.log
2025-02-13 20:19:54 [0ms] [Warning] [omni.kit.app.plugin] No crash reporter present, dumps uploading isn't available.
2025-02-13 20:19:55 [873ms] [Warning] [omni.usd_config.extension] Enable omni.materialx.libs extension to use MaterialX
2025-02-13 20:19:55 [1,603ms] [Warning] [omni.isaac.dynamic_control] omni.isaac.dynamic_control is deprecated as of Isaac Sim 4.5. No action is needed from end-users.

|---------------------------------------------------------------------------------------------|
| Driver Version: 535.129.03    | Graphics API: Vulkan
|=============================================================================================|
| GPU | Name                             | Active | LDA | GPU Memory | Vendor-ID | LUID       |
|     |                                  |        |     |            | Device-ID | UUID       |
|     |                                  |        |     |            | Bus-ID    |            |
|---------------------------------------------------------------------------------------------|
| 0   | NVIDIA GeForce RTX 4090          | Yes: 0 |     | 24564   MB | 10de      | 0          |
|     |                                  |        |     |            | 2684      | 86a573cf.. |
|     |                                  |        |     |            | 1         |            |
|=============================================================================================|
| OS: 20.04.6 LTS (Focal Fossa) ubuntu, Version: 20.04.6, Kernel: 5.15.0-105-generic
| XServer Vendor: The X.Org Foundation, XServer Version: 12013000 (1.20.13.0)
| Processor: 13th Gen Intel(R) Core(TM) i9-13900KF
| Cores: 8 | Logical Cores: 16
|---------------------------------------------------------------------------------------------|
| Total Memory (MB): 31824 | Free Memory: 23648
| Total Page/Swap (MB): 2047 | Free Page/Swap: 17592186044414
|---------------------------------------------------------------------------------------------|
[INFO]: Parsing configuration from: isaaclab_tasks.manager_based.manipulation.reach.config.franka.joint_pos_env_cfg:FrankaReachEnvCfg
[INFO]: Parsing configuration from: isaaclab_tasks.manager_based.manipulation.reach.config.franka.agents.rsl_rl_ppo_cfg:FrankaReachPPORunnerCfg
[INFO] Logging experiment in directory: /home/yuhao/IsaacLab/logs/rsl_rl/franka_reach
Exact experiment name requested from command line: 2025-02-13_15-20-01
Setting seed: 42
[INFO]: Base environment:
	Environment device    : cuda:0
	Environment seed      : 42
	Physics step-size     : 0.016666666666666666
	Rendering step-size   : 0.03333333333333333
	Environment step-size : 0.03333333333333333
./isaaclab.sh: line 274: 46359 Killed                  ${python_exe} "$@"
```

Thank you very much for your help!

RandomOakForest, 2025-02-15:
Thanks for following up. The team will investigate.

achao2013, 2025-03-04:
The same problem. What is a recommended machine configuration that won't experience similar problems @RandomOakForest

StafaH, 2026-09-25:
Closing this github issue as it is now stale. IsaacLab 3.0 has migrated to a completely new stack, with a focus on pip installable packages and easier installation. Please feel free to re-open the issue if there are issues with the new installation with uv on HPC.

closed 2026-09-25 by StafaH