isaac-sim/IsaacLab #1419: How to install OpenUSD via NVIDIA Omniverse? Needed for making custom assets (or combining custom assets)

meghbhalerao, 2024-11-13:
I download the zip file here https://developer.nvidia.com/downloads/USD/usd_binaries/24.08/usd.py310.manylinux_2_35_x86_64.usdview.release@0.24.08-gl.7245+v24.08.59992d21.zip as mentioned on this page https://developer.nvidia.com/usd#section-getting-started
[image]
I download the zip file for linux and unzip it (I am using ubuntu 24) - but there are no instructions on how to proceed after unzipping the zip files? Could someone help me with this?
After unzipping I get the following files in the unzipped dir - 
```
bin                pip-packages
BUILD_INFO         plugin
CHANGELOG.md       pxrConfig.cmake
cmake              python
include            README.md
launcher.toml      resources
lib                scripts
libraries          share
LICENSE            tests
PACKAGE-DEPS.yaml  THIRD-PARTY.md
PACKAGE-INFO.yaml  usd.py310.manylinux_2_35_x86_64.usdview.release@0.24.08-gl.7245+v24.08.59992d21.zip
PACKAGE-LICENSES
```

It would be great to get some help on how to proceed after this point! Or if someone could point me to the right forum to get this question answered.
Thanks!

RandomOakForest, 2024-11-16:
Thank you for posting this. We will tag it for improving our documentation in an upcoming release.

matthewtrepte, 2026-10-19:
glad you got this to work. since its about a 3rd party tool, i'm closing.

closed 2026-10-19 by matthewtrepte