isaac-sim/IsaacLab #1423: [Bug Report] MDP IMU observation functions do not work with ObservationManager

diracdelta7, 2024-11-14:
### Describe the bug

When using the `imu_lin_acc` and `imu_ang_vel` observation functions from **mdp/observations.py**, I encounter the following error:

```
RuntimeError: The update function must be called before the data buffers are accessed the first time.
```

### Steps to reproduce

1. Add an IMU sensor to a `ManagerBasedRLEnv` instance. For example, in `isaaclab_locomotion/tasks/locomotion/velocity/velocity_env_cfg.py`, add the following configurations:

In `MySceneCfg`:
```python
imu_robot_base: ImuCfg = ImuCfg(
    prim_path="{ENV_REGEX_NS}/Robot/base",
)
```

In `ObservationsCfg`:
```python
base_imu_lin_acc = ObsTerm(
    func=mdp.imu_lin_acc,
    params={"asset_cfg": SceneEntityCfg("imu_robot_base")},
)
```

2. Run the training script with a velocity environment, such as the Anymal-D environment:
```sh
python source/standalone/workflows/rsl_rl/train.py --task=Isaac-Velocity-Flat-Anymal-D-v0 --headless
```

Upon running the script, the following error occurs:
```
Error executing job with overrides: []
Traceback (most recent call last):
  File "/home/xin/rl/IsaacLab/source/extensions/omni.isaac.lab_tasks/omni/isaac/lab_tasks/utils/hydra.py", line 99, in hydra_main
    func(env_cfg, agent_cfg, *args, **kwargs)
  File "/home/xin/rl/IsaacLab/source/standalone/workflows/rsl_rl/train.py", line 101, in main
    env = gym.make(args_cli.task, cfg=env_cfg, render_mode="rgb_array" if args_cli.video else None)
  File "/home/xin/anaconda3/envs/isaaclab/lib/python3.10/site-packages/gymnasium/envs/registration.py", line 802, in make
    env = env_creator(**env_spec_kwargs)
  File "/home/xin/rl/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/envs/manager_based_rl_env.py", line 75, in __init__
    super().__init__(cfg=cfg)
  File "/home/xin/rl/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/envs/manager_based_env.py", line 138, in __init__
    self.load_managers()
  File "/home/xin/rl/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/envs/manager_based_rl_env.py", line 115, in load_managers
    super().load_managers()
  File "/home/xin/rl/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/envs/manager_based_env.py", line 215, in load_managers
    self.observation_manager = ObservationManager(self.cfg.observations, self)
  File "/home/xin/rl/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/managers/observation_manager.py", line 75, in __init__
    super().__init__(cfg, env)
  File "/home/xin/rl/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/managers/manager_base.py", line 131, in __init__
    self._prepare_terms()
  File "/home/xin/rl/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/managers/observation_manager.py", line 343, in _prepare_terms
    obs_dims = tuple(term_cfg.func(self._env, **term_cfg.params).shape)
  File "/home/xin/rl/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/envs/mdp/observations.py", line 231, in imu_lin_acc
    return asset.data.lin_acc_b
  File "/home/xin/rl/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/sensors/imu/imu.py", line 80, in data
    self._update_outdated_buffers()
  File "/home/xin/rl/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/sensors/sensor_base.py", line 289, in _update_outdated_buffers
    self._update_buffers_impl(outdated_env_ids)
  File "/home/xin/rl/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/sensors/imu/imu.py", line 146, in _update_buffers_impl
    raise RuntimeError(
RuntimeError: The update function must be called before the data buffers are accessed the first time.
```

It appears that `ObservationManager` calls `mdp.imu_lin_acc` before `IMU::update()` is triggered, and `IMU._dt` is undefined, which is required in `IMU::_update_buffers_impl`.

### System Info

Describe the characteristic of your environment:

- Commit: https://github.com/isaac-sim/IsaacLab/commit/84b2d2d8849e86752aa5a948d5f818d8430eec5a
- Isaac Sim Version: 4.2.0-rc.17+release.15988.c99988b1.gl
- OS: Ubuntu 20.04
- GPU: RTX A6000
- CUDA: 11.8
- GPU Driver: 535.183.01

### Checklist

- [x] I have verified that no similar issue exists in the repository.
- [x] I have confirmed that this issue is unrelated to Isaac Sim functionality and is specific to the repository.

### Acceptance Criteria

Add the criteria for which this task is considered done. If not known at issue creation time, you can add this once the issue is assigned.

- [ ] Fix the bug or provide recommended usage of the imu observation functions to avoid this error.

RandomOakForest, 2024-11-16:
Thank you for posting this. Could you try the [latest release](https://github.com/isaac-sim/IsaacLab/releases/tag/v1.2.0) instead of that commit? Thanks!

diracdelta7, 2024-11-16:
Thank you for your reply! The IMU sensor and its configuration were added in [this commit](https://github.com/isaac-sim/IsaacLab/commit/be5260375edaa3bad2e3cc0cd87ec4ac3ac01d45), which is after the [latest release](https://github.com/isaac-sim/IsaacLab/releases/tag/v1.2.0) you referenced. I'm currently using the latest `main` branch.

diegoaldarondo-fauna, 2024-11-18:
I ran into this issue as well. I've temporarily resolved it by calling `env.scene.update` (which eventually calls the imu update method) after resetting the simulation context and before loading the managers in the `manager_based_env` init method.

```
        if builtins.ISAAC_LAUNCHED_FROM_TERMINAL is False:
            print(
                "[INFO]: Starting the simulation. This may take a few seconds. Please wait..."
            )
            with Timer("[INFO]: Time taken for simulation start", "simulation_start"):
                self.sim.reset()
            # add timeline event to load managers
            self.scene.update(dt=self.physics_dt) # Add this line
            self.load_managers()
```

RandomOakForest, 2025-01-24:
Following up on this. A workaround has been merged recently to fix this issue. Let us know if you still see this error. Thank you.

Mayankm96, 2025-02-06:
@jtigue-bdai I think the fix proposed here should be the more correct way to go since we have other sensors involved as well and the issue may creep up again.

The sim.reset() already does a "dummy" physics step inside, so it should be okay to do a scene update.

Since eventually we do a reset call for all the environments for training/playing, the internal counters will be set back to zero so it shouldn't affect the sensor updates.

```python
     if builtins.ISAAC_LAUNCHED_FROM_TERMINAL is False:
            print(
                "[INFO]: Starting the simulation. This may take a few seconds. Please wait..."
            )
            with Timer("[INFO]: Time taken for simulation start", "simulation_start"):
                self.sim.reset()
            # add timeline event to load managers
            self.scene.update(dt=self.physics_dt) # Add this line
            self.load_managers()
```

giulioturrisi, 2025-02-07:
For the **direct environment**, i'm adding in **direct_rl_env.py** the same self.scene.update(dt=self.physics_dt) line as above. But I'm getting the following error:

`
Traceback (most recent call last):
  File "/home/gturrisi/isaaclab_ws_home/IsaacLab/source/isaaclab_tasks/isaaclab_tasks/utils/hydra.py", line 101, in hydra_main
    func(env_cfg, agent_cfg, *args, **kwargs)
  File "/home/gturrisi/isaaclab_ws_home/quadruped_rl_collection/scripts/rsl_rl/train.py", line 105, in main
    env = gym.make(args_cli.task, cfg=env_cfg, render_mode="rgb_array" if args_cli.video else None)
  File "/home/gturrisi/mambaforge/envs/isaaclab_env/lib/python3.10/site-packages/gymnasium/envs/registration.py", line 740, in make
    env = env_creator(**env_spec_kwargs)
  File "/home/gturrisi/isaaclab_ws_home/quadruped_rl_collection/source/quadruped_rl_collection/quadruped_rl_collection/tasks/locomotion/locomotion_env.py", line 33, in __init__
    super().__init__(cfg, render_mode, **kwargs)
  File "/home/gturrisi/isaaclab_ws_home/IsaacLab/source/isaaclab/isaaclab/envs/direct_rl_env.py", line 141, in __init__
    self.scene.update(dt=self.physics_dt) # Add this line
  File "/home/gturrisi/isaaclab_ws_home/IsaacLab/source/isaaclab/isaaclab/scene/interactive_scene.py", line 504, in update
    articulation.update(dt)
  File "/home/gturrisi/isaaclab_ws_home/IsaacLab/source/isaaclab/isaaclab/assets/articulation/articulation.py", line 202, in update
    self._data.update(dt)
AttributeError: 'Articulation' object has no attribute '_data'
`

jtigue-bdai, 2025-02-11:
@giulioturrisi can you provide a reproduction of you issue using a direct_rl_env?

giulioturrisi, 2025-02-13:
Yes, here is the code (just a fork of isaaclab):
https://github.com/giulioturrisi/IsaacLab-test/tree/test_imu

Check the latest commit https://github.com/giulioturrisi/IsaacLab-test/commit/29a2a468b7f3850412207cf8a49e1253dcf5d870 to check where I added the IMU

To launch the direct env where I added the IMU, use 
`./isaaclab.sh -p scripts/reinforcement_learning/rsl_rl/train.py --task=Isaac-Velocity-Flat-Anymal-C-Direct-v0 --headless`

giulioturrisi, 2025-02-13:
Found out the problem. I was adding self.scene.update(dt=self.physics_dt) before self.sim.reset(). In your pull request you did in the opposite order, and everything now works.

PR #1809 (2025-02-14): Adds scene update to ManagerBasedEnv and DirectRLEnv initialization
# Description

This PR fixes execution of imu (and other sensor) based observation by updating the scene at the initialization of ManagerBasedEnv and DirectRLEnv. This ensures the Imu._dt (and any other scene buffers) are set before extracting Imu.data.

Fixes #1423 

## Type of change

- Bug fix (non-breaking change which fixes an issue)

## Screenshots

Please attach before and after screenshots of the change if applicable.

## Checklist

- [x] I have run the [`pre-commit` checks](https://pre-commit.com/) with `./isaaclab.sh [... cut]
commit 157c6b74 (2025-02-14): Adds scene update to ManagerBasedEnv and DirectRLEnv initialization (#1809)

# Description

This PR fixes execution of imu (and other sensor) based observation by
updating the scene at the initialization of ManagerBasedEnv and
DirectRLEnv. This ensures the Imu._dt (and any other scene buffers) are
set before extracting Imu.data.

Fixes #1423 

## Type of change

- Bug fix (non-breaking change which fixes an issue)

## Screenshots

Please attach before and after screenshots of the change if applicable.

## Checklist

- [x] I have run the [`pre-commit` checks](https://pre-commit.com/) with
`./is [... cut]
closed 2025-02-14 by kellyguo11