isaac-sim/IsaacLab #1425: How can I get rendering and scene update faster?

hojae-io, 2024-11-15:
Hi, I'm looking for a way to increase the rendering and simulation speed.

1. First, I'm trying to train my policy with `sim_dt=0.001` and `decimation=10`, unlike usual examples where `sim_dt=0.005` and `decimation=4`.
This cause my training time very slow since the physics stepping loop in `manager_based_rl_env.py` takes longer time.
I observed that the main bottleneck is `self.scene.write_data_to_sim()` and `self.scene.update()`.
Is there any way that I can make this function faster?

2. Also, the `self.sim.render()` takes 0.01 seconds which indicates that the rendering cannot exceed 100Hz by the render function itself. I think this is too slow. Is there any way that I can reduce this render time in exchange of rendering quality? Wondering if there's any solution other than just skipping some frames.

Thanks!

Please refer to the bottom related issues:
https://github.com/isaac-sim/IsaacLab/discussions/1157
https://github.com/isaac-sim/IsaacLab/issues/379

RandomOakForest, 2024-11-16:
Thanks for posting this. While we get a more specific answer, have you tried the [Isaac Sim speed-up guidelines here](https://docs.omniverse.nvidia.com/isaacsim/latest/reference_material/speedup_cheat_sheet.html)?

RandomOakForest, 2024-12-06:
I hope the guidelines posted worked for you. Please post this into a *discussion* for follow up if you need further help. I'll close this for now. Thanks.

closed 2024-12-06 by RandomOakForest