isaac-sim/IsaacLab #1430: [Proposal] Add `replace_cylinders_with_capsules` option in Urdf converters

ZiwenZhuang, 2024-11-18:
### Proposal

Adding `replace_cylinders_with_capsules` in the urdf converters configuration class.

### Motivation

Previously in IsaacGym/legged_gym, `replace_cylinders_with_capsules` option can make the simulation faster and more stable. However, when radius is a lot greater than cylinders' length, the collision shape will be extremely different.

Please consider adding this option in the implementation of `_get_urdf_import_config`

### Additional context

I've checked that this does not need any other implementations, just adding two lines of code.

1. In `_get_urdf_import_config` of `UrdfConverter` in `omni.isaac.lab/omni/Isaac/lab/sim/converters/urdf_converters.py`
[image]

2. In `UrdfConverterCfg` in `omni.isaac.lab/omni/Isaac/lab/sim/converters/urdf_converters_cfg.py`
[image]

### Checklist

- [x] I have checked that there is no similar issue in the repo (**required**)

RandomOakForest, 2024-11-18:
Thank you for posting this. The team will review as a possible enhancement in an upcoming release.

Mayankm96, 2025-03-31:
This flag is now exposed in the commit 9bb4c42. Closing the issue.

closed 2025-03-31 by Mayankm96