isaac-sim/IsaacLab #1436: [Question] Robot instability and flying behavior in Isaac Lab with Ridgeback and Franka

JinMinChan, 2024-11-19:
I am using Isaac Sim 4.2.0 and Isaac Lab 1.2.0 to train a task involving turning a valve with the ridgeback_franka robot. The valve.usd was created manually, and I have confirmed that it works correctly (successful operation with Franka alone). However, when replacing Franka with ridgeback_franka, I encountered issues.

For the ridgeback_franka, I used the ridgeback_franka.usd provided by Omniverse. The script follows the direct approach, and the basic robot configuration was taken as is from ridgeback_franka.py. Additionally, I compared my robot's state with the one in source/extensions/omni.isaac.lab/test/assets/check_ridgeback_franka.py and verified that they are identical.

However, when running my code, the robot tilts toward the Franka side immediately after initialization and eventually flies away. I would greatly appreciate any assistance regarding this issue. As .py uploads are not allowed, I am attaching the script as a .txt file instead.

Thank you always for your support.

[my_code.txt](https://github.com/user-attachments/files/17812293/my_code.txt)
[image]

SuomiKP31, 2024-11-19:
Try to configure joint frictions in your asset config. They default to the values set within the asset which are most of the time zero.
This will result in rigidbodies gaining energy from self collisions and ground collisions, then eventually they fly off to infinity.

JinMinChan, 2024-11-20:
> Can you show an image of your scene in its initial state with the collision bounding boxes visible?
> 
> Can you also check the mass of your valve rigid bodies relative to the mass of the franka robot's rigid bodies? If these are dramatically difference it can cause a lot of simulation issues.

[image]

Thank you for your input. Here is the information you requested:

Initial State Collision Shapes
The attached image shows the collision shapes in their initial state.

Mass Values
Robot
word: 0.0001
dummy_base_x: 0.0001
dummy_base_y: 0.0001
base_link: 165.304
arm_mount_link: 0.0001
panda_link0: 4.0
panda_link1: 3.0
panda_link2: 3.0
panda_link3: 2.0
panda_link4: 2.0
panda_link5: 2.0
panda_link6: 1.5
panda_link7: 0.5
panda_hand: 0.55833
panda_leftfinger: 0.01406
panda_rightfinger: 0.01406
endeffector: 0.0001
front_laser: 0.13
rear_laser: 0.13
Valve (Original)
base: Autocomputed
valve_body: Autocomputed
valve_handle: Autocomputed
Valve (Modified)
I updated the valve's mass values as follows:

base: 15
valve_body: 40
valve_handle: 7.5
However, the issue persists even after applying these changes.

JinMinChan, 2024-11-20:
> Try to configure joint frictions in your asset config. They default to the values set within the asset which are most of the time zero. This will result in rigidbodies gaining energy from self collisions and ground collisions, then eventually they fly off to infinity.

Thank you for your input. As you mentioned, I found that the Joint Friction for all joints was set to 0. I tested increasing it incrementally from 0.1 to 1, but the issue of the robot flying away still persists. However, I did observe that increasing the friction reduced the extent of the flying behavior.

RandomOakForest, 2025-01-27:
Following up on this issue, have you found any resolutions? is this still persistent with the latest Isaac Lab? If you still need further help we may move this to a _discussion_ to follow up by the team. Thanks.

closed 2025-02-11 by RandomOakForest

wangrp1997, 2025-05-10:
@JinMinChan,  Hi, I have the same problem. Have you solved it now?

RandomOakForest, 2025-05-10:
Thanks for following up. It would be best if you open a new discussion with an update of where you're at. Then the team and others can follow up. Thanks.

SuddenCucumber, 2026-01-05:
similar problem here, i am setting up RL on a quadruped robot and it keeps rotating around its z-axis even when i manually modify the function _apply_action(self) in the environment class to set all joints (all actions) to 0. The issue still persists when i use rot[] to change the quaternion of the robot in the robot config. my suspicion is that isaaclab sees the root link as a rotatory joint and tries to rotate it, but i am not sure where to fix that (urdf and usd are out of the question because i looked at them and there is no joint that the root link could rotate around.